DreamSourceLab/DSView #47: Load and Save configuration

bvernoux, 2015-03-16:
Load and Save configuration including decoder used, channel name and all settings.
As today DSLogic is not usable for multiple protocol tests as we cannot load/save configuration and we shall always reconfigure all from scratch.

bvernoux, 2015-04-12:
Any news on this vital feature ? (as today I'm using ScanaPlus for my tests because DSLogic lacks saving/loading configuration ...)
Does DSLogic is an abandonned project (about 1 year without any update !!!) ?
* No update since Oct, 15 2014 in https://github.com/DreamSourceLab/DSLogic
* No update since Jul 8, 2014 in https://github.com/DreamSourceLab/DSLogic-hdl
* No update since Aug 11, 2014 in https://github.com/DreamSourceLab/DSLogic-fw

bvernoux, 2015-05-24:
It seems fixed in latest release https://github.com/DreamSourceLab/DSView/releases/tag/0.93
But the binary have not been released and I'm a bit lazy to install all the things to build it myself.
When it is planned to release this version ?
In http://www.dreamsourcelab.com/download.html ?

DreamSourceLab, 2015-06-29:
v0.94 version added session load/store function.
You can try the new version.
Thanks.

bvernoux, 2015-06-29:
I have just installed http://www.dreamsourcelab.com/download/DSView(Beta)_v0.9.4.1_setup.exe on Win7 SP1 64Bits.
I have loaded this file:
http://hydrabus.com/HydraNFC_spi_ss_tests.dsl
Then I configure the SPI:
[image]
All is ok
* Then I save it as Session (File->Session->Store) to file HydraNFC_spi_ss_tests.dsc
And the file is saved with 0bytes so it does not work.
* Then I save it as "DSView Data" (File -> Save...) to a new file HydraNFC_spi_ss_tests_.dsl
If you reload HydraNFC_spi_ss_tests_.dsl it is corrupted:
[image]

DreamSourceLab, 2015-12-14:
Fixed in v0.95 version.

closed 2015-12-14 by DreamSourceLab

bvernoux, 2015-12-14:
It is not fixed