DreamSourceLab/DSView #51: Use common code formatting standard for the source tree

anatol, 2015-05-17:
It would be easier to read and support the code if it used common formatting standards.

There are several tools that help to maintain formatting and one of them is [clang-format](http://clang.llvm.org/docs/ClangFormat.html). Here is an example how to use it:

`find . \( -name \*.cpp -or -name \*.h \) -exec clang-format -i -style='{BasedOnStyle: llvm, ColumnLimit: 100}' '{}' \;`

It is possible to create formatting that your prefer and save to `.clang-format` file.

DreamSourceLab, 2015-05-21:
Thank you very much for this suggestion.
We will try it.

anatol, 2015-05-21:
Thanks. Note that it makes sense to format `DSView` folder only. Avoid formatting libsigrok4DSL, libusb to make future merges easier.