DreamSourceLab/DSView #52: Win7 64bits issue with DSView(Beta)_v0.931_setup.exe

bvernoux, 2015-05-24:
I have just installed latest DSView from
http://www.dreamsourcelab.com/download/DSView(Beta)_v0.931_setup.exe

1) I load an old file trf7970a_nfm_spi_new_fpga_bitstream_ok.dsl all is ok.
See file http://hydrabus.com/trf7970a_nfm_spi_new_fpga_bitstream_ok.dsl
2) I save it to a new file trf7970a_nfm_spi_new_ok.dsl all is ok.
See file http://hydrabus.com/trf7970a_nfm_spi_new_ok.dsl
3) Now if i try to open trf7970a_nfm_spi_new_ok.dsl I obtain error

[image]

DreamSourceLab, 2015-06-29:
It may be a bug when save virtual driver's data.
Thanks.

bvernoux, 2015-06-29:
See my comment https://github.com/DreamSourceLab/DSView/issues/47
This issue is not fixed with http://www.dreamsourcelab.com/download/DSView(Beta)_v0.9.4.1_setup.exe

DreamSourceLab, 2015-12-14:
Fixed in v0.95 version.

closed 2015-12-14 by DreamSourceLab