DreamSourceLab/DSView #58: Project does not compile under Ubuntu Vivid (15.04)

ticpu, 2015-07-09:
Fails with
```
Building CXX object CMakeFiles/DSView.dir/pv/mainwindow.cpp.o
/home/jerome/GIT/DSLogic/DSView/pv/mainwindow.cpp: In member function ‘bool pv::MainWindow::store_session(QString)’:
/home/jerome/GIT/DSLogic/DSView/pv/mainwindow.cpp:742:77: error: conversion from ‘uint64_t {aka long unsigned int}’ to ‘const QVariant’ is ambiguous
             s_obj["vdiv"] = QJsonValue::fromVariant(dsoSig->get_vDialValue());
```

Then fails with:
```
Building CXX object CMakeFiles/DSView.dir/pv/data/decode/annotation.cpp.o
/home/jerome/GIT/DSLogic/DSView/pv/data/decode/annotation.cpp: In constructor ‘pv::data::decode::Annotation::Annotation(const srd_proto_data*)’:
/home/jerome/GIT/DSLogic/DSView/pv/data/decode/annotation.cpp:43:17: error: ‘const struct srd_proto_data_annotation’ has no member named ‘ann_class’
  _format = pda->ann_class;
                 ^
CMakeFiles/DSView.dir/build.make:1695: recipe for target 'CMakeFiles/DSView.dir/pv/data/decode/annotation.cpp.o' failed
```

wd5gnr, 2015-07-11:
Casting the return value inside the fromVariant calls to type quint64 solves the first problem. I am still trying to solve the 2nd problem.

wd5gnr, 2015-07-11:
The second one is the ann_class vs ann_format problem so I assume we are conflicting on H files somewhere. Will see if it works but it does get you past that compile.

epileftric, 2015-07-25:
> Casting the return value inside the fromVariant calls to type quint64 solves the first problem. I am still trying to solve the 2nd problem.

I tried that and didn't work for me. I directly made a QString from the number and avoided the QJsonValue constructor

> The second one is the ann_class vs ann_format problem so I assume we are conflicting on H files somewhere. Will see if it works but it does get you past that compile.

There's a patch in the arch AUR tha simply changes that for:

    - _format = pda->ann_class;
    + _format = pda->ann_format;

anatol, 2015-08-30:
The issues should be fixed in 'develop' branch now. Please check it.

DreamSourceLab, 2015-12-14:
fixed in v0.95 version.

closed 2015-12-14 by DreamSourceLab