DreamSourceLab/DSView #59: Can't compile: error in mainwindow.cpp

epileftric, 2015-07-24:
There was some error when compiling `inpv/mainwindow.cpp` wrote the following patch:

        diff --git a/DSView/pv/mainwindow.cpp b/DSView/pv/mainwindow.cpp
    --- mainwindow.cpp.orig	2015-07-24 19:00:36.621046681 -0300
    +++ mainwindow.cpp	2015-07-24 19:01:13.749582511 -0300
    @@ -739,8 +739,8 @@
             s_obj["strigger"] = s->get_trig();
             boost::shared_ptr<view::DsoSignal> dsoSig;
             if (dsoSig = dynamic_pointer_cast<view::DsoSignal>(s)) {
    -            s_obj["vdiv"] = QJsonValue::fromVariant(dsoSig->get_vDialValue());
    -            s_obj["vfactor"] = QJsonValue::fromVariant(dsoSig->get_factor());
    +            s_obj["vdiv"] = QString("%1").arg(dsoSig->get_vDialValue());
    +            s_obj["vfactor"] = QString("%1").arg(dsoSig->get_factor());
                 s_obj["coupling"] = dsoSig->get_acCoupling();
                 s_obj["trigValue"] = dsoSig->get_trigRate();
                 s_obj["zeroPos"] = dsoSig->get_zeroRate();

deepfryed, 2015-10-10:
looks like this has been fixed in tree/develop.

epileftric, 2015-10-10:
Indeed. I recompiled today and didn't come up with this issue, still many others on run time.

DreamSourceLab, 2015-12-14:
fixed in v0.95 version.

closed 2015-12-14 by DreamSourceLab