DreamSourceLab/DSView #60: Slow protocol Decoder

ganeshkumarsw, 2015-08-10:
When sampled UART signal for 28M Samples @10MHz, the protocol analyser takes more than 2mins to decode the data which is too sluggish in my opinion.

bvernoux, 2015-08-10:
Could you attach the file in to check that ? (and also to be used later for comparison & check improvment)

ganeshkumarsw, 2015-08-10:
First of all, thanks for coming back. The problem can be simulated very
easily even in demo mode of application.  Set samples to 16M @ 10Mhz and
then add protocol decoder on any two channels for uart, you will see the
result for yourself. I will also attach the the screen shot from my side.

ganeshkumarsw, 2015-08-10:
[image]

ganeshkumarsw, 2015-08-11:
I have also attached an video to show how long the decoder takes to decode

ganeshkumarsw, 2015-08-11:
I did the same testing that I performed on DSlogic over pulseview, but here the pulse view is even more faster than Saleae in terms of protocol decoding,

ganeshkumarsw, 2015-08-12:
Use UART decoder @115200 for the attached log under DSView, it takes around 6mins until its done decoding. Its too long time, I am exhausted, so quick diagnosis is not possible with this tool then but I am happy about the Hardware but software needs lots of improvement in terms of decoder.

Link for download:
https://www.dropbox.com/s/ygjysg2xonxb9k1/DSView_TestData.dsl?dl=0

deepfryed, 2015-10-10:
Yeah I'd love the protocol decoder optimization - currently for large samples it takes way too long to be useful debugging tool.

ganeshkumarsw, 2015-10-10:
@deepfryed : Any release planned for this optimization?

DreamSourceLab, 2017-06-06:
Improved since v0.96

closed 2017-06-06 by DreamSourceLab