DreamSourceLab/DSView #61: Sampling in us, ms and sec

ganeshkumarsw, 2015-08-10:
In DS logic software, sampling shall be made available in the options of us, ms and sec for the convenience instead of samples in mere numbers so it easy for the user to select what he wants rather than back calculating samples to sec or ms or us.

This similar option is available in saleae logic software.

ganeshkumarsw, 2015-08-10:
[image]

DreamSourceLab, 2018-07-09:
Supported on DSView v0.99
[image]

DreamSourceLab, 2018-07-20:
@flowswitch 
When used with external clock, you just need to select a right sample rate.

flowswitch, 2018-07-24:
@DreamSourceLab 
What sample rate is "right" for aperiodic/gated clock then?
Ok, even for periodic, but gated case (i.e. SD bus with idle or TSIF bus), sample rate selected to match the clock - how long is "1 second" of sample depth?

DreamSourceLab, 2018-07-24:
@flowswitch
We think it's unsolvable. The settings of sample duration and sample rate only decide the total sample count(sample duration*sample rate) when using external clock. It means capture will stop when sample count reached.

closed 2019-09-10 by DreamSourceLab