DreamSourceLab/DSView #62: make in libsigrok4DSL error : ../../libsigrok.h:27:18: fatal error: glib.h: No such file or directory

mintisan, 2015-08-15:
`./configure` is right :
```
checking for strstr... no
checking for strtol... no
checking for strtoul... no
checking for strtoull... no
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating version.h
config.status: creating hardware/Makefile
config.status: creating hardware/demo/Makefile
config.status: creating hardware/common/Makefile
config.status: creating hardware/DSL/Makefile
config.status: creating input/Makefile
config.status: creating output/Makefile
config.status: creating libsigrok4DSL.pc
config.status: creating tests/Makefile
config.status: creating config.h
config.status: config.h is unchanged
config.status: executing depfiles commands
config.status: executing libtool commands

libsigrok configuration summary:

  - Package version (major.minor.micro):    0.2.0
  - Library version (current:revision:age): 1:2:0
  - Prefix: /usr/local
  - Building on: i686-pc-linux-gnu
  - Building for: i686-pc-linux-gnu

Detected libraries:

  - glib-2.0 >= 2.32.0: yes (2.40.2)
  - libzip >= 0.10: yes (0.10.1)
  - libserialport >= 0.1.0: no
  - libusb-1.0 >= 1.0.9: no
  - libftdi >= 0.16: no
  - libudev >= 151: no
  - alsa >= 1.0: no
  - check >= 0.9.4: no

Enabled hardware drivers:

  - demo............................ yes
  - DreamSourceLab.................. yes

```

but when I `make`, the error comes:
```
make  all-recursive
Making all in hardware
Making all in demo
  CC       libsigrok4DSL_hw_demo_la-demo.lo
In file included from demo.c:32:0:
../../libsigrok.h:27:18: fatal error: glib.h: No such file or directory
 #include <glib.h>
                  ^
compilation terminated.
make[3]: *** [libsigrok4DSL_hw_demo_la-demo.lo] Error 1
make[2]: *** [all-recursive] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
````

And I am sure I installed glib :
```
 sudo apt-get install glib2.0-dev
正在读取软件包列表... 完成
正在分析软件包的依赖关系树       
正在读取状态信息... 完成       
Note, selecting 'libglib2.0-dev' for regex 'glib2.0-dev'
libglib2.0-dev is already the newest version.
libglib2.0-dev 被设置为手动安装。
升级了 0 个软件包，新安装了 0 个软件包，要卸载 0 个软件包，有 490 个软件包未被升级。
```

maaraneasi, 2015-08-15:
Yes - same here... I am not able to build on opensuse 13.2 even ale glibh requirements are installed and in place.

mintisan, 2015-08-15:
@maaraneasi ,I just solve this problem.

1.make sure instal all the packges below, don't paste with `\`, because terminal will end with it
```
 $ sudo apt-get install git-core gcc g++ make cmake autoconf automake libtool pkg-config libglib2.0-dev libzip-dev libudev-dev libusb-1.0-0-dev python3-dev qt5-default libboost-dev libboost-test-dev libboost-thread-dev libboost-system-dev libboost-filesystem-dev check
```

2.if u encounter error below, you can `sudo apt-get update` & re-try install packages again

```
E: 无法下载 http://archive.ubuntu.com/ubuntu/pool/main/s/systemd/udev_204-5ubuntu20.12_i386.deb  404  Not Found [IP: 91.189.91.23 80]

E: 无法下载 http://archive.ubuntu.com/ubuntu/pool/main/s/systemd/libudev1_204-5ubuntu20.12_i386.deb  404  Not Found [IP: 91.189.91.23 80]

E: 无法下载 http://archive.ubuntu.com/ubuntu/pool/main/s/systemd/libudev-dev_204-5ubuntu20.12_i386.deb  404  Not Found [IP: 91.189.92.201 80]

E: 有几个软件包无法下载，您可以运行 apt-get update 或者加上 --fix-missing 的选项再试试？
```

3.then **re-configure makefile** & follow the INSTALL is just ok
```
Step3: Building
 
 $ cd libsigrok4DSL
 $ ./autogen.sh
 $ ./configure
 $ make
 $ sudo make install
 $ cd ..

 $ git clone git://sigrok.org/libsigrokdecode
 $ cd libsigrokdecode
 $ ./autogen.sh
 $ ./configure
 $ make
 $ sudo make install
 $ cd ..

 $ cd DSView
 $ cmake .
 $ make
 $ sudo make install
```

4.launch DSView with `sudo`,or u will encounter error below

```
./DSView
```

[image]

5.the last success screen-capture , `sudo ./DSView`, u will listen a **click** sound

[image]

FYI

mintisan, 2015-08-15:
@maaraneasi BTW, I build it on Linux Mint 17 x86.32bit

closed 2015-08-15 by mintisan