DreamSourceLab/DSView #63: Problem: sampling 50MHz signal at different rate

ganeshkumarsw, 2015-08-21:
Here with I have attached dsl files, which holds the sampling of 50Mhz SPI CLK signal with setting of 16M samples @100MHz, 200MHz and 400MHz, its quite interesting to know that each file measures different SPI clock rate but at least @200Mhz works fine but even then it varies in duty cycle from 50% to 75%, its weird behaviour. 

I feel the hardware is stable enough till 25MHz signal beyond that its not promising, if you have different opinion, kindly share. 

For the sake of attachment I have renamed the file to png, kindly save and rename it to .dsl

**Is it a firmware issue or software issue?**

[image]

[image]

[image]

bvernoux, 2015-08-27:
Pictures in link cannot be displayed could you fix the links ? (could you re-upload the pictures)

ganeshkumarsw, 2015-08-28:
please save the image and rename to dsl file, the you can use ds view to
check this

int0, 2015-12-03:
Was this ever fixed? I still see same behavior on higher frequencies. I guess I was too optimistic to sample 133Mhz SPI signal.

ganeshkumarsw, 2015-12-03:
Its not fixed. I am disappointed with features offered by this tool.

1. Slow decoding with more no of samples.
2. Hardly 83ms sampling time with 16 mega sample for signals of 200Mhz.

DreamSourceLab, 2015-12-14:
fixed in v0.95 version.

closed 2015-12-14 by DreamSourceLab