DreamSourceLab/DSView #65: Capture and plot analog signal at "Logic Analyzer" screen

anatol, 2015-11-11:
Currently the Logic Analyzer app allows to capture digital signal only.

Sometimes it is very useful to capture and compare both digital signal (e.g. I2C) and analog signal (e.g. speaker output) at the same screen. Saleae software allows it. It would be great if DSView was able to do the same.

shumtravi, 2017-10-30:
Me voting for that.

liuzheng1990, 2018-12-28:
Me, too. I bought the analyzer in order to measure the voltage time series. Now I felt it's useless for me. Can this be done by modifying the software, or do we have to add anything to the hardware itself?

DreamSourceLab, 2018-12-29:
Logic analyzer can only process digital signal. If you want to measure the voltage, you should choose DSCope(our oscilloscope product, also run on DSView software).