DreamSourceLab/DSView #66: Save (Load) session does not work => 0 byte file

bvernoux, 2015-12-14:
It is not fixed again !! see  Load and Save configuration #47
Are you doing some tests before closing issues ??

I have loaded this file:
http://hydrabus.com/HydraNFC_spi_ss_tests.dsl
Then I configure the SPI:
[image]
Warning loading old files change some channels !!
* Then I save it as Session (File->Session->Store) to file HydraNFC_spi_ss_tests.dsc
And the file is saved with 0bytes so it does not work.
* Then I save it as "DSView Data" (File -> Save...) to a new file HydraNFC_spi_ss_tests_.dsl
If you reload HydraNFC_spi_ss_tests_.dsl it is ok

Conlusion:
Load/Store for "Session" is broken even with latest release.
Warning loading old files change some channels !!
You have all files to reproduce this problem.

bvernoux, 2016-03-26:
Any news on this important feature to support ?
And also a way to export (even in simple ASCII text file or csv) decoder values for post analysis...

DreamSourceLab, 2017-06-06:
Please try 0.97 version

closed 2017-06-06 by DreamSourceLab

bvernoux, 2017-06-06:
The session (decoders) are now saved/loaded correctly