DreamSourceLab/DSView #68: Protocol decoder  and QT issue on an Ubuntu linux compilation.

teyssieuman, 2015-12-20:
I compiled the current sources (0.95) and I experiment the following issue.

As soon as I request the "decoder" tab, the following message appears in the shell.

```
QWidget::paintEngine: Should no longer be called
QPainter::begin: Paint device returned engine == 0, type: 1
QPainter::setRenderHint: Painter must be active to set rendering hints
QPainter::setRenderHint: Painter must be active to set rendering hints
```

I don't know if this issue is linked to the following one but: I am not able to use any decoder under ubuntu. (it works under my windows computer) ...

DreamSourceLab, 2015-12-23:
Is there any warnings when you compile the code? What you posted should not be the cause.

teyssieuman, 2015-12-23:
Here are the logs:
```
fabrice@assus1:~/Bureau/DSView-master/DSView/DSView$ cmake .
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.28") 
-- The C compiler identification is GNU 5.2.1
-- The CXX compiler identification is GNU 5.2.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- checking for modules 'libsigrok4DSL >= 0.2.0;libusb-1.0 >= 1.0.16;libsigrokdecode>=0.3.0'
--   found libsigrok4DSL , version 0.2.0
--   found libusb-1.0 , version 1.0.18
--   found libsigrokdecode, version 0.4.0-git-033e7d4
-- Using Qt5
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   filesystem
--   system
--   thread
-- Found libusb-1.0:
--  - Includes: /usr/local/include
--  - Libraries: /usr/local/lib/libusb-1.0.so
-- Configuring done
-- Generating done
-- Build files have been written to: /home/fabrice/Bureau/DSView-master/DSView/DSView
```

and then :

```
fabrice@assus1:~/Bureau/DSView-master/DSView/DSView$ make
[  1%] Generating qrc_DSView.cpp
[  1%] Generating pv/moc_mainwindow.cpp
[  2%] Generating pv/moc_sigsession.cpp
[  2%] Generating pv/moc_storesession.cpp
[  3%] Generating pv/device/moc_devinst.cpp
[  4%] Generating pv/dialogs/moc_about.cpp
[  4%] Generating pv/dialogs/moc_deviceoptions.cpp
[  5%] Generating pv/dialogs/moc_search.cpp
[  6%] Generating pv/dialogs/moc_storeprogress.cpp
[  6%] Generating pv/dialogs/moc_streamoptions.cpp
[  7%] Generating pv/dialogs/moc_waitingdialog.cpp
[  8%] Generating pv/dialogs/moc_dsomeasure.cpp
[  8%] Generating pv/dock/moc_dsotriggerdock.cpp
[  9%] Generating pv/dock/moc_measuredock.cpp
[ 10%] Generating pv/dock/moc_searchdock.cpp
[ 10%] Generating pv/dock/moc_triggerdock.cpp
[ 11%] Generating pv/prop/moc_bool.cpp
[ 12%] Generating pv/prop/moc_double.cpp
[ 12%] Generating pv/prop/moc_enum.cpp
[ 13%] Generating pv/prop/moc_int.cpp
[ 14%] Generating pv/prop/moc_property.cpp
[... 110 lines cut ...]
[ 86%] Building CXX object CMakeFiles/DSView.dir/pv/prop/moc_property.cpp.o
[ 86%] Building CXX object CMakeFiles/DSView.dir/pv/prop/moc_string.cpp.o
[ 87%] Building CXX object CMakeFiles/DSView.dir/pv/toolbars/moc_filebar.cpp.o
[ 88%] Building CXX object CMakeFiles/DSView.dir/pv/toolbars/moc_logobar.cpp.o
[ 88%] Building CXX object CMakeFiles/DSView.dir/pv/toolbars/moc_samplingbar.cpp.o
[ 89%] Building CXX object CMakeFiles/DSView.dir/pv/toolbars/moc_trigbar.cpp.o
[ 90%] Building CXX object CMakeFiles/DSView.dir/pv/view/moc_cursor.cpp.o
[ 90%] Building CXX object CMakeFiles/DSView.dir/pv/view/moc_devmode.cpp.o
[ 91%] Building CXX object CMakeFiles/DSView.dir/pv/view/moc_header.cpp.o
[ 92%] Building CXX object CMakeFiles/DSView.dir/pv/view/moc_ruler.cpp.o
[ 92%] Building CXX object CMakeFiles/DSView.dir/pv/view/moc_selectableitem.cpp.o
[ 93%] Building CXX object CMakeFiles/DSView.dir/pv/view/moc_timemarker.cpp.o
[ 94%] Building CXX object CMakeFiles/DSView.dir/pv/view/moc_trace.cpp.o
[ 94%] Building CXX object CMakeFiles/DSView.dir/pv/view/moc_view.cpp.o
[ 95%] Building CXX object CMakeFiles/DSView.dir/pv/view/moc_viewport.cpp.o
[ 96%] Building CXX object CMakeFiles/DSView.dir/pv/widgets/moc_fakelineedit.cpp.o
[ 96%] Building CXX object CMakeFiles/DSView.dir/pv/dock/moc_protocoldock.cpp.o
[ 97%] Building CXX object CMakeFiles/DSView.dir/pv/data/moc_decoderstack.cpp.o
[ 98%] Building CXX object CMakeFiles/DSView.dir/pv/view/moc_decodetrace.cpp.o
[ 98%] Building CXX object CMakeFiles/DSView.dir/pv/widgets/moc_decodergroupbox.cpp.o
[ 99%] Building CXX object CMakeFiles/DSView.dir/pv/widgets/moc_decodermenu.cpp.o
[100%] Building CXX object CMakeFiles/DSView.dir/qrc_DSView.cpp.o
Linking CXX executable DSView
[100%] Built target DSView
```

teyssieuman, 2015-12-23:
There is a log of warning when I compile "libsigrok4DSL":

```
make  all-recursive
Making all in hardware
Making all in demo
  CC       libsigrok4DSL_hw_demo_la-demo.lo
In file included from demo.c:32:0:
../../libsigrok.h:154:77: warning: 'struct sr_dev_inst' declared inside parameter list
 typedef int (*sr_receive_data_callback_t)(int fd, int revents, const struct sr_dev_inst *sdi);
                                                                             ^
../../libsigrok.h:154:77: warning: its scope is only this definition or declaration, which is probably not what you want
demo.c: In function 'hw_dev_mode_list':
demo.c:381:18: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for(i = 0; i < ARRAY_SIZE(mode_list); i++) {
                  ^
demo.c: In function 'samples_generator':
demo.c:743:17: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
         pre_buf = sinx;
                 ^
demo.c:746:17: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
         pre_buf = sqrx;
                 ^
demo.c:749:17: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
         pre_buf = trix;
                 ^
demo.c:752:17: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
         pre_buf = sawx;
                 ^
demo.c:755:17: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
         pre_buf = ranx;
                 ^
demo.c:758:17: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
         pre_buf = sinx;
                 ^
demo.c: In function 'receive_data':
demo.c:875:27: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
             for (i = 0; i < sending_now; i++) {
                           ^
demo.c:941:29: warning: assignment from incompatible pointer type [-Wincompatible-pointer-types]
                 analog.data = devc->buf;
                             ^
demo.c:966:33: warning: passing argument 1 of 'hw_dev_acquisition_stop' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
         hw_dev_acquisition_stop(sdi, NULL);
                                 ^
demo.c:289:12: note: expected 'struct sr_dev_inst *' but argument is of type 'const struct sr_dev_inst *'
 static int hw_dev_acquisition_stop(struct sr_dev_inst *sdi, void *cb_data);
            ^
demo.c:842:35: warning: unused variable 'expected_samplenum' [-Wunused-variable]
  static uint64_t samples_to_send, expected_samplenum, sending_now;
                                   ^
demo.c: In function 'hw_dev_acquisition_start':
demo.c:998:30: warning: implicit declaration of function 'ds_trigger_get_mask0' [-Wimplicit-function-declaration]
         devc->trigger_mask = ds_trigger_get_mask0(TriggerStages);
                              ^
demo.c:999:31: warning: implicit declaration of function 'ds_trigger_get_value0' [-Wimplicit-function-declaration]
         devc->trigger_value = ds_trigger_get_value0(TriggerStages);
                               ^
demo.c:1000:30: warning: implicit declaration of function 'ds_trigger_get_edge0' [-Wimplicit-function-declaration]
         devc->trigger_edge = ds_trigger_get_edge0(TriggerStages);
                              ^
demo.c:1031:18: warning: passing argument 4 of 'sr_session_source_add_channel' from incompatible pointer type [-Wincompatible-pointer-types]
             100, receive_data, sdi);
                  ^
In file included from ../../libsigrok.h:1096:0,
                 from demo.c:32:
../../proto.h:123:12: note: expected 'sr_receive_data_callback_t {aka int (*)(int,  int,  const struct sr_dev_inst *)}' but argument is of type 'int (*)(int,  int,  const struct sr_dev_inst *)'
 SR_API int sr_session_source_add_channel(GIOChannel *channel, int events,
            ^
demo.c: At top level:
[... 656 lines cut ...]
                                              ^
trigger.c:307:46: warning: suggest parentheses around comparison in operand of '|' [-Wparentheses]
                  trigger->trigger1[stage][i] == 'C');
                                              ^
  CC       version.lo
In file included from version.c:21:0:
libsigrok.h:154:77: warning: 'struct sr_dev_inst' declared inside parameter list
 typedef int (*sr_receive_data_callback_t)(int fd, int revents, const struct sr_dev_inst *sdi);
                                                                             ^
libsigrok.h:154:77: warning: its scope is only this definition or declaration, which is probably not what you want
  CC       error.lo
In file included from error.c:21:0:
libsigrok.h:154:77: warning: 'struct sr_dev_inst' declared inside parameter list
 typedef int (*sr_receive_data_callback_t)(int fd, int revents, const struct sr_dev_inst *sdi);
                                                                             ^
libsigrok.h:154:77: warning: its scope is only this definition or declaration, which is probably not what you want
  CC       std.lo
In file included from std.c:22:0:
libsigrok.h:154:77: warning: 'struct sr_dev_inst' declared inside parameter list
 typedef int (*sr_receive_data_callback_t)(int fd, int revents, const struct sr_dev_inst *sdi);
                                                                             ^
libsigrok.h:154:77: warning: its scope is only this definition or declaration, which is probably not what you want
  CCLD     libsigrok4DSL.la
ar: `u' modifier ignored since `D' is the default (see `U')
```

teyssieuman, 2015-12-29:
I made a video to demonstrate this issue: 
https://youtu.be/bjITW4hfILs

Please ask me if you want me to practice some more experiment ...

DreamSourceLab, 2017-06-06:
Please try 0.97 version

closed 2017-06-06 by DreamSourceLab