DreamSourceLab/DSView #74: Segfault on linux after Configure FPGA using "DSCope.bin"

leonardoromor, 2016-01-09:
Hello everyone, I'm using arch linux I just bought DSCope and tried to install DSView on 
my laptop, sadly after installing the software the oscilloscope was working only on windows and mac os.

So I dig in, and I've found the following bug:

if ((fw = g_fopen(filename, "rb")) == NULL) { 

the fucntion g_fopen starts a segfault:

sr: Device came back after 1843ms.
sr: Configure FPGA using /usr/share/dsview/DSCope.bin
Segmentation fault (core dumped)

substituting g_fopen with just fopen did the trick and now I can normally use the oscilloscope,

I'm using glib2-2.46.2-2

I still don't understand whose fault is (maybe mine?),
 thank you,

Leo

zittix, 2016-01-09:
Just replace all the g_fopen calls by fopen and this will solve your problem.

leonardoromor, 2016-01-09:
Yes, I have already solved the problem in this way (as I already wrote), but I don't understand why this happens... also, I believe that after paying 200$ for a working software and hardware you should fix it for me (or not?)

zittix, 2016-01-09:
First I have nothing to do with DreamSourceLab and I totally agree that we deserve a better software, although the hardware is quite good.

Concerning the bug, I read somewhere that mixing f*** functions and g_f**** functions in the code is known to create some problems with certain version of glib. I cannot find back the post mentioning that though.

leonardoromor, 2016-01-09:
Thank you very much, I'm sorry for the last message I was a bit frustrated for the afternoon
lost debugging, I've also found in the documentation of glib  a line that says exactly what you are saying https://developer.gnome.org/glib/stable/glib-File-Utilities.html, 

I hope they will fix the problem, thanks again,

Leo

DreamSourceLab, 2017-06-06:
Fixed on 0.97 version

closed 2017-06-06 by DreamSourceLab