DreamSourceLab/DSView #76: Zero Adjustment for Oscilloscope does not work.

PePaPossum, 2016-01-27:
I am having a problem with the zero adjustment function when trying to use the oscilloscope on DSView v0.9.5.1 for Windows (same fault on 3 different computers).

The zero adjustment is not completing (meaning I cannot use the oscilloscope function) - it just has a circle with moving dots and says waiting. I have left it for upwards of half and hour and it is still saying waiting. I do not have any scope probes connected to the oscilloscope. I have tried running DSView as an administrator but it has not made any difference.

Can you please let me know what is causing this?

Thanks,
Daniel

AkosLukacs, 2016-01-27:
Try restarting you computer. Seriously, this happened to me: #69

kriskizlyk, 2016-03-12:
I am having the same issue.  Even after a reboot still stays on Waiting, waited over an hour...

Original Kickstarter version with scope add on, Win 7 64, Scope Probes Unplugged or Plugged in,  v0.9.5.1(Beta, please tell me why they release it on Beta with no previous version download links available).

DreamSourceLab, 2017-06-06:
Fixed on 0.97 version

closed 2017-06-06 by DreamSourceLab