DreamSourceLab/DSView #80: Linux Use Channels 0-3 (400MHz) not possible (V0.9.5)

De-Backer, 2016-04-28:
radio button does not select:
Use Channels 0-7 (Max 200MHz)
Use Channels 0-3 (Max 400MHz)
in linux (opensuse leap 42.1 64bit)

De-Backer, 2016-04-28:
devinst.h
line 120-123
signals:
	void config_changed();

is not connect to a SLOT

DreamSourceLab, 2016-05-03:
what the color of hardware indicator?
Did you try it in root user?

De-Backer, 2016-05-04:
what the color of hardware indicator? => green
And runing on 16 Channels 100MHz
Blinking on retrieving data.
Did you try it in root user? => yes

devsaurus, 2017-01-15:
I have a similar issue on current build 0.9.6 (https://github.com/DreamSourceLab/DSView/commit/acc14cf0791cb8710733a064c1c77600a2d0c404) when running DSView as a regular user:
The channel mode cannot be changed via the GUI.
I can change it by editing `~/.local/share/DSView/DSLogic0.dsc` though.

I haven't observed any further restrictions yet, but this is a bit inconvenient.

The radio buttons do work as expected when running the application as root.

vatrat, 2017-04-06:
This works for me on Fedora, with the device DSLogicPro. Should this be closed?

cire831, 2017-06-06:
why was this close?

Not able to reproduce?

vatrat, 2017-06-07:
Hmmmmm, not sure if closing was justified. They just updated to 0.97. Try to download and see if this bug still exists. I don't have windows.

DreamSourceLab, 2017-06-07:
We can't reproduce this issue.
Please help to check on 0.97 version.
If no response, we will close it.

closed 2017-08-19 by DreamSourceLab