DreamSourceLab/DSView #87: Hangs at 100% on Arch Linux - libusb event handling on multiple threads

FuzzyCheese, 2016-11-21:
Hangs on my Arch system and on a fresh Antergos install but works on Ubuntu fresh install.

Symptom: Stays at 100% completion and UI is frozen and must be killed when used with DSLogic hardware.

Crude fix:
Comment out libusb event handling in receive_data function in libsigrok4DSL/hardware/DSL/dslogic.c around line 2365.

--- a/DSView-master/libsigrok4DSL/hardware/DSL/dslogic.c
+++ b/DSView-master/libsigrok4DSL/hardware/DSL/dslogic.c
@@ -2380,11 +2380,11 @@
usb = sdi->conn;

tv.tv_sec = tv.tv_usec = 0;
- libusb_handle_events_timeout_completed(drvc->sr_ctx->libusb_ctx, &tv, &completed);
+//fix libusb_handle_events_timeout_completed(drvc->sr_ctx->libusb_ctx, &tv, &completed);

if (devc->status == DSL_FINISH) {
- if (libusb_try_lock_events(drvc->sr_ctx->libusb_ctx) == 0) {
- if (libusb_event_handling_ok(drvc->sr_ctx->libusb_ctx)) {
+//fix if (libusb_try_lock_events(drvc->sr_ctx->libusb_ctx) == 0) {
+//fix if (libusb_event_handling_ok(drvc->sr_ctx->libusb_ctx)) {
/* Stop GPIF acquisition */
usb = ((struct sr_dev_inst *)devc->cb_data)->conn;
if ((ret = command_stop_acquisition (usb->devhdl)) != SR_OK)
@@ -2393,9 +2393,9 @@
sr_info("%s: Sent acquisition stop command!", __func__);

remove_sources(devc);
- }
- libusb_unlock_events(drvc->sr_ctx->libusb_ctx);
- }
+//fix }
+//fix libusb_unlock_events(drvc->sr_ctx->libusb_ctx);
+//fix }
}

return TRUE;

Details:
The hang occurs because of usblib event handling with multiple threads.?.?
The following line(113) in libsigrok4DSL/hardware/command.c does not return at the end of final transfer:
ret = libusb_control_transfer(devhdl, LIBUSB_REQUEST_TYPE_VENDOR |
LIBUSB_ENDPOINT_OUT, CMD_START, 0x0000, 0x0000,
(unsigned char *)&cmd, sizeof(cmd), 3000);

Used "export LIBUSB_DEBUG=4" to turn on libusb debug and "DSView -l 5" for application logging and it looks like the event is returned to another thread causing the current thread to wait on line 113.

Log(trimmed):
sr: finish_acquisition: send SR_DF_END packet
sr: session: bus: Received SR_DF_END packet.
[ 3.219458] [000042bf] libusb: debug [libusb_alloc_transfer] transfer 0x7f94fc000ed0
[ 3.219463] [000042bf] libusb: debug [libusb_submit_transfer] transfer 0x7f94fc000ed0
[ 3.219466] [000042bf] libusb: debug [add_to_flying_list] arm timerfd for timeout in 3000ms (first in line)
[ 3.219491] [000042bf] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling
...repeats many times....
[ 6.298252] [000042be] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling
[ 6.298330] [000042be] libusb: debug [libusb_cancel_transfer] transfer 0x7f94fc000ed0
[ 6.298338] [000042be] libusb: debug [discard_urbs] URB not found --> assuming ready to be reaped
[ 6.298347] [000042be] libusb: debug [libusb_cancel_transfer] cancel transfer failed error -5
[ 6.298357] [000042be] libusb: warning [handle_timeout] async cancel failed -5 errno=22
[ 6.398431] [000042be] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling

vatrat, 2016-12-30:
I was about to make an issue for this, but on Fedora 25. I'll try your fix and update this post.

FuzzyCheese, 2016-12-31:
The DSLogic indicator is orange(dim red and dim green) when it has no firmware. DSView detects the DSLogic hardware, uploads the firmware and the indicator turns solid green. It blinks green when sending data. Then steady green when idle. I have only seen red a few times when something was wrong.

vatrat, 2016-12-31:
Oh, okay. Since I switched all of my devices to Arch and Fedora (from Ubuntu), I didn't have a working device to test with. I think I have seen the light be red until the device is connected to DSView, though? I'll have to check it when I have time.

Anyone have any idea what's causing this? Knowledge of libusb or what the difference between Ubuntu installations and other installations is?

yaakovgamliel, 2017-01-02:
I'm having exactly the same issue its working fine (after placing udev rules in the proper place) in Ubuntu, but in my main system which is runing Arch the thing just hangs forever, I'll try to PR a fix if I find one soon

kkrizka, 2017-01-21:
I am experiencing the same problem on Gentoo. I played around the libsigrok4DSL source code today. If I understand how libusb works, I think that @FuzzyCheese's patch is the final fix. The ``receive_data`` function gets the event handling lock via ``libusb_try_lock_events`` and then calls ``command_stop_acquisition`` to finish the acquisition. However ``command_stop_acquisition`` performs a synchronous USB command in the same thread, which requires event handing. However no thread is doing the event handing since the ``recieve_data`` put a lock on it and is being blocked by the ``command_stop_acquisition`` function. 

Does this make sense? If I am right, this would mean that this should never works. But it does on Ubuntu and Windows.  This is my first time with libusb, so please excuse me if I misunderstood something.

@yaakovgamliel, what version of Ubuntu did you have it working? I would like to compare the differences in libusb used by the distributions

yaakovgamliel, 2017-01-22:
I'm using 16.04 LTS

kkrizka, 2017-01-22:
Everything seems consistent now. Ubuntu 16.04 is using libusb-1.0.20 while I have libusb-1.0.21. One thing they changed was to make the event handing lock non-recursive, so the same thread cannot lock the mutex twice: https://github.com/libusb/libusb/commit/a5302ff86db391e6797a32693c242ddade5a09c2

yaakovgamliel, 2017-02-03:
Sad to see this project abandoned by the maintainers, I understand that they are a very small team but maybe if they engaged a bit more with the community, we could together make this better, The DS is as very cool device after all and the only complain that raises always is the software behind it.

vatrat, 2017-03-30:
Hey, I'm doing some work with the DSLogic and I've forked the code. I'll probably make some other changes as well, but would anyone like for me to make a pull request with this fix?

fenrig, 2017-03-31:
You could try a pull request. If they don't respond we will resort to using
your fork.

yaakovgamliel, 2017-03-31:
@vatrat Try the PR, in any case I'll be happy to try out your fork and contribute to it

fenrig, 2017-03-31:
@vatrat: :+1: keep up the good work, if they dont care we'll just maintain a fork ;) thats what opensource is for

vatrat, 2017-04-05:
Okay, just added this tempfix and submitted a pull request. Now I'll make a branch of my fork and merge this and pull request #92.

cire831, 2017-04-05:
I've talked to the developers and they say they are actively working on the  code.  

But I agree, there isn't much evidence of that.

we should agree where the new "master" is going to be located.  We do want to be careful so we can track any changes that occur on the main DreamSource code base.  I know how to do that.

I'd be happy to host it on one of my main areas..

Probably TinyProd, <github.com/TinyProd>  would be the best, if you guys want.

I'm actively using DSLogicPro on my projects as well.  Very nice unit.

vatrat, 2017-04-05:
Really, you've talked to the devs? What kind of stuff are they doing?

cire831, 2017-04-05:
not sure if it was a dev or not.

i bought more probes and I asked what was up with the development tree.  They told me they were working on it.  But it is a mystery to me what exactly there are doing.  It isn't visible.

vatrat, 2017-05-27:
Since PR #96 was merged, should this issue be closed?

DreamSourceLab, 2017-05-30:
Thanks for all of your guys.
We have released a commit to fix this issue.
Could you please have a try? 
If no problem, we will close this issue.

closed 2017-06-06 by DreamSourceLab