DreamSourceLab/DSView #94: DSView窗口丢失

LyCrystals, 2017-04-09:
当连接外置显示器的时候将窗口移动到外置显示器上。关闭应用。再断开外置显示器。再打开应用的时候会出现窗口丢失。程序是开着的，但是窗口找不回来。
操作系统win10;
AcerV5-591G

vatrat, 2017-04-13:
Google translate:

> # DSView window is missing
> Moves the window to an external monitor when connecting an external monitor. Close the application. And then disconnect the external display. And then open the application will appear when the window is lost. The program is open, but the window is not coming back.
Operating system win10;
AcerV5-591G

XieyangLy, 2017-04-18:
i'm sorry to say this it doesn't work at all.the way ,you mentioned,can be uesd to other application such as firefox,chrome,but not DSView.
while i hold the shift key,right click on the DSView icon in the taskbar,there is no shortcut menu display.
the DSView version:v0.9.6 simple chinese version;
the OS:win 10 version,1703

DreamSourceLab, 2017-06-06:
Fixed on 0.97 version

closed 2017-06-06 by DreamSourceLab

vatrat, 2017-06-07:
I don't think this should be closed until there is a response

XieyangLy, 2017-06-08:
thanks, the new version solve the problem.