DreamSourceLab/DSView #95: crash button found

lpdunwell, 2017-04-12:
this button should not exist
[image]

vatrat, 2017-04-13:
What OS, version, etc. are you using? I'll try to reproduce this later. Does DSView crash when you click this button?

lpdunwell, 2017-04-13:
win7x64
yes

vatrat, 2017-04-13:
Okay, thank you. Are you using a version of DSView that you downloaded from the site? If so, is it 0.96?

lpdunwell, 2017-04-13:
it's the recent one from your site.
quickest fix would be to delete that button. it doesn't have any purpose
anyway.

also, there's some spelling errors. i haven't yet had time to look at the
actual code, but i will (long week end) and submit that.

cheers

vatrat, 2017-04-13:
1. Not my site. I'm just a user.
2. There is a PR that fixes at least a few spelling errors, PR #92.

I'll try to look at the code today. I'm not too familiar with the UI code, but it's Qt, so it shouldn't be that bad. I'll see if I can just delete the button.

vatrat, 2017-04-17:
Okay, I finally got to my computer and tried this out. When I click the button on Linux, with the same "1:I2C" setting, it just brings up a dialog box that says:

> Message
> 
> Protocol Analyzer
> 
> No Protocol Analyzer to delete!
> 
> [OK]

I'll still try to remove the button, but it's worth noting that it doesn't crash for me.

DreamSourceLab, 2017-06-06:
We don't reproduce this issue on 0.97 version.
Could you help to try it?
If no problem, we will close the issue.

closed 2017-08-19 by DreamSourceLab