DreamSourceLab/DSView #100: make issues lots warnings

rolfschumacher, 2017-05-21:
I followed INSTALL on Debian Jessie.
I got so many warnings that I cannot believe that the software will do what it should.
For now, I stopped to use the software on Linux.

What did I wrong?

[DSViewMakeError.txt](https://github.com/DreamSourceLab/DSView/files/1016738/DSViewMakeError.txt)
[DSViewMake.txt](https://github.com/DreamSourceLab/DSView/files/1016739/DSViewMake.txt)

vatrat, 2017-05-21:
This is not an issue. Warnings are fine. The compilation is using the setting "-Wall", which shows all warnings. Ideally, someone would go and fix all the warnings to keep the warning output clear, but it won't cause any problems if you don't.

vatrat, 2017-05-27:
Hey, just wanted to say that I'm working on this now. As well, the developers have released two commits in the last few days that fix some warnings.

DreamSourceLab, 2017-06-06:
Please try 0.97 version

closed 2017-08-19 by DreamSourceLab