DreamSourceLab/DSView #104: sigrok scope API

lromor, 2017-06-28:
Hello, 
I have a DScope and I would like to ask you if you have an API to control the scope without the GUI.

Which classes should I look at?  Is there a class with simple methods to achieve what I need? 
Thanks,

-l

vatrat, 2017-06-28:
Hey, just a user here. I don't think they have anything like that, but look
in the "hardware" folder in the repository. Or, maybe you can build your
own version of the application and strip down the functions in the GUI for
use as an API.

lromor, 2017-06-28:
Thanks for the answer, 
I think they should have some abstraction level for the data acquisition, instead of reading many lines of code (as I already started doing) I think it would be much easier if they could just tell me at which file to stop.

vatrat, 2017-06-28:
Probably not just one file. And, remember, this program is a fork from
Sigrok. It takes a lot of work to give something an API.

lromor, 2017-06-28:
well, it's still a lib, and it has an API by itself, just asking for some howto, but while digging between the files I found that the headers proto.h, backend.c sigrok.h do contain some useful notes on how to use the library.

I think I will start from there and probably copy the usage from DSView. But the main functions are well listed in proto.h

vatrat, 2017-06-28:
Okay, thanks for the info. What are you going to do with it?

lromor, 2017-06-28:
I need this in order to setup some data acquisition. I need to capture k samples when trigger starts. Right now the lib seems to be nicely done, so for today I managed to link the device and initialized the drivers. 

So now, if I connect the device shows: 

`sr: Sanity-checking all drivers.
sr: Sanity-checking all input modules.
sr: Sanity-checking all output modules.
sr: hwdriver: Initializing driver 'virtual-demo'.
sr: hwdriver: Initializing driver 'DSLogic'.
sr: hwdriver: Initializing driver 'DSCope'.
sr: hwdriver: Scan of 'virtual-demo' found 1 devices.
sr: hwdriver: Scan of 'DSLogic' found 0 devices.
sr: Found an DSCope device.
sr: hwdriver: Scan of 'DSCope' found 1 devices.
`

vatrat, 2017-06-29:
Neat! Would you mind putting your work in a github repo, if you haven't already? It'd be a nice starting point for other projects.