DreamSourceLab/DSView #109: CAN FD decoder

kanji2000, 2017-09-05:
I really need 'CAN FD' decoder.
Could you please support it.

CZZ187827, 2022-09-15:
> 

我根据Can的写了一个CanFD的。

wdim0, 2022-09-20:
CAN-FD is more and more common nowadays. I'd like to use DSView to debug my CAN-FD project. It would be great to have CAN-FD decoder in DSView.

abougouffa, 2022-09-20:
@wdim0 @CZZ187827 @1101459202 

I've integrated some of `sigrok`s decoders into DSView from a while, I didn't get the time to test them all, the new CAN decoder supports CAN-FD.

I've just opened a PR, if you want to test the new decoder, you can clone from my fork https://github.com/abougouffa/DSView/tree/feat/decoders

wdim0, 2022-09-25:
@abougouffa

I've cloned your fork and tried it, but there's an error:
[image]

I've also checked the latest version of ".../libsigrokdecode4DSL/decoders/can/pd.py" on sigrok web
http://sigrok.org/gitweb/?p=libsigrokdecode.git;a=tree;f=decoders/can;h=a9b52f173e2ca7df101087af1da8e8e5e117402f;hb=HEAD

but the file has the same content (except one line at start in comment with copyright) and it produces the same error.
I'm afraid we have to wait until somebody with good "python + sigrok decoder architecture" skills will fix that.

Thanks anyway!

abougouffa, 2022-09-25:
@wdim0 You can try to replace these lines:

https://github.com/abougouffa/DSView/blob/feat/decoders/libsigrokdecode4DSL/decoders/can/pd.py#L517-L520

With:

```python
if (self.matched & (0b1 << 1)):
    self.dom_edge_seen()
if (self.matched & (0b1 << 0)):
    self.handle_bit(can_rx)
```

It is the `libsigrokdecode4dsl` which is not up-to-date, compared with the upstream `libsigrokdecode`. I don't understand why they made their own fork, which is constantly far behind the upstream one, instead of depending on the upstream lib directly. There is a very good proposition in #428, but the response from DreamSourceLab's developer was kind of hostile!

wdim0, 2022-09-27:
@abougouffa
Yep! That's working!!
Thank you so much [abougouffa](https://github.com/abougouffa)

CAN-FD frame:
[image]

RoopaPalyegar, 2023-07-18:
Hi, I am new here. I need to decode CAN-FD data. Critical stage of the proejct. 
Can anyone offer any help? 
I downloaded the zipped folder from "https://github.com/sigrokproject/libsigrokdecode"
Replaced "decoders" folder in my laptop with the 'decoders' folder from the downloaded zipped folder.
Replaced the 4 lines as suggested by @wdim0 
But I get "Decoder Reported an error"

[image]

abougouffa, 2023-07-18:
Hello @RoopaPalyegar 

AFAK, [CAN-FD](https://github.com/DreamSourceLab/DSView/tree/master/libsigrokdecode4DSL/decoders/can-fd) is supported in the last 1.3.0 version. This hack was proposed for an older version.

wanckl, 2025-04-15:
@abougouffa Hi, Dose CAN-FD decoder has some improvements later, I'm using DSview 1.3.2, It works for general frams, but can't detect error frames or something like below    

[image]

abougouffa, 2025-04-15:
Hi @wanckl 

I don't think so... If you have some examples of CAN-FD's features that aren't working correctly, you can try to open a separate ticket.