DreamSourceLab/DSView #115: DSView crashes on OS X Sierra when i try to run fft.

shumtravi, 2017-10-25:
DSView 0.9.8 crashes on OS X Sierra when i try to run fft. At moment i press button OK in Fft parameters menu DSView crashes.

> Process:               DSView [72860]
> Path:                  /Applications/DSView.app/Contents/MacOS/DSView
> Identifier:            com.DreamSourceLab.DSView
> Version:               ???
> Code Type:             X86-64 (Native)
> Parent Process:        ??? [1]
> Responsible:           DSView [72860]
> User ID:               501
> 
> Date/Time:             2017-10-26 01:30:04.534 +0300
> OS Version:            Mac OS X 10.12.6 (16G29)
> Report Version:        12
> 
> System Integrity Protection: enabled
> 
> Crashed Thread:        0  Dispatch queue: com.apple.main-thread
> 
> Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
> Exception Codes:       0x0000000000000001, 0x0000000000000000
> Exception Note:        EXC_CORPSE_NOTIFY
> 
> Termination Signal:    Illegal instruction: 4
> Termination Reason:    Namespace SIGNAL, Code 0x4
> Terminating Process:   exc handler [0]
> 
> Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
> 0   com.DreamSourceLab.DSView     	0x000000010fee812e 0x10fda0000 + 1343790

shumtravi, 2017-11-19:
Process:               DSView [94032]
Path:                  /Applications/DSView.app/Contents/MacOS/DSView
Identifier:            com.DreamSourceLab.DSView
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           DSView [94032]
User ID:               501

Date/Time:             2017-11-19 12:26:28.427 +0300
OS Version:            Mac OS X 10.13.1 (17B48)
Report Version:        12
Anonymous UUID:        

Sleep/Wake UUID:       

Time Awake Since Boot: 19000 seconds
Time Since Wake:       1100 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Illegal instruction: 4
Termination Reason:    Namespace SIGNAL, Code 0x4
Terminating Process:   exc handler [0]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.DreamSourceLab.DSView     	0x00000001012cf12e 0x101187000 + 1343790

Thread 1:: OGL Profiler
0   libsystem_kernel.dylib        	0x00007fff580d1e76 mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff580d1390 mach_msg + 60
2   com.apple.opengl              	0x00007fff3ac1299f glcDebugListener + 303
3   libsystem_pthread.dylib       	0x00007fff582156c1 _pthread_body + 340
4   libsystem_pthread.dylib       	0x00007fff5821556d _pthread_start + 377
5   libsystem_pthread.dylib       	0x00007fff58214c5d thread_start + 13

Thread 2:: org.libusb.device-hotplug
0   libsystem_kernel.dylib        	0x00007fff580d1e76 mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff580d1390 mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff309aa5d5 __CFRunLoopServiceMachPort + 341
3   com.apple.CoreFoundation      	0x00007fff309a9927 __CFRunLoopRun + 1783
4   com.apple.CoreFoundation      	0x00007fff309a8fa3 CFRunLoopRunSpecific + 483
5   com.apple.CoreFoundation      	0x00007fff309e7993 CFRunLoopRun + 99
6   libusb-1.0.0.dylib            	0x0000000101763249 0x101758000 + 45641
7   libsystem_pthread.dylib       	0x00007fff582156c1 _pthread_body + 340
8   libsystem_pthread.dylib       	0x00007fff5821556d _pthread_start + 377
9   libsystem_pthread.dylib       	0x00007fff58214c5d thread_start + 13

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff580dae7e __psynch_cvwait + 10
1   libsystem_pthread.dylib       	0x00007fff58216662 _pthread_cond_wait + 732
2   com.DreamSourceLab.DSView     	0x0000000101295249 0x101187000 + 1106505
3   com.DreamSourceLab.DSView     	0x000000010129583e boost::this_thread::hidden::sleep_until(timespec const&) + 78
4   com.DreamSourceLab.DSView     	0x0000000101198b40 pv::SigSession::hotplug_proc(boost::function<void (QString)>) + 400
5   com.DreamSourceLab.DSView     	0x00000001011a3306 0x101187000 + 115462
6   com.DreamSourceLab.DSView     	0x00000001011a3203 0x101187000 + 115203
7   com.DreamSourceLab.DSView     	0x00000001011a3194 0x101187000 + 115092
8   com.DreamSourceLab.DSView     	0x00000001012947c4 0x101187000 + 1103812
9   libsystem_pthread.dylib       	0x00007fff582156c1 _pthread_body + 340
10  libsystem_pthread.dylib       	0x00007fff5821556d _pthread_start + 377
11  libsystem_pthread.dylib       	0x00007fff58214c5d thread_start + 13

Thread 4:: com.apple.NSEventThread
0   libsystem_kernel.dylib        	0x00007fff580d1e76 mach_msg_trap + 10
[... 28 lines cut ...]
1   libsystem_pthread.dylib       	0x00007fff5821526f _pthread_wqthread + 1552
2   libsystem_pthread.dylib       	0x00007fff58214c4d start_wqthread + 13

Thread 8:
0   libsystem_pthread.dylib       	0x00007fff58214c40 start_wqthread + 0
1   ???                           	0x00000000c2000000 0 + 3254779904

Thread 9:
0   libsystem_kernel.dylib        	0x00007fff580db6da __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff5821526f _pthread_wqthread + 1552
2   libsystem_pthread.dylib       	0x00007fff58214c4d start_wqthread + 13

Thread 10:
0   libsystem_pthread.dylib       	0x00007fff58214c40 start_wqthread + 0

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x00007ff516c5cf00  rcx: 0x0000000000000000  rdx: 0x00007ff516c60460
  rdi: 0x0000000101745080  rsi: 0x000000010203ba00  rbp: 0x00007ff516c60380  rsp: 0x00007ffeeea6fca0
   r8: 0x0000000000000000   r9: 0x00007ff516c60370  r10: 0x000007ff516c6043  r11: 0x0000000000000001
  r12: 0x00000001015c8f10  r13: 0x0000000000000004  r14: 0x0000000000000001  r15: 0x0000000000000001
  rip: 0x00000001012cf12e  rfl: 0x0000000000010297  cr2: 0x000000010134010e
  
Logical CPU:     0
Error Code:      0x00000000
Trap Number:     6

shumtravi, 2017-11-19:
Does anybody have worked Math/FFT in OSX?

DreamSourceLab, 2018-07-08:
@shumtravi 
We have checked the FFT function of DSView 0.98 on MacOS 10.13.
It works well. 
Additionally, you can also try our new version:
[http://dreamsourcelab.com/download/DSView-v0.99.dmg](url)

DreamSourceLab, 2018-07-09:
[image]

DreamSourceLab, 2018-07-18:
@outthereandaway 
We also checked v0.99 on Mac OS 10.13. It works well.
We will update system to 10.13.6 and have a try again.
By the way, you can try v0.98, but we think it will have the same result.

qunanan, 2018-08-23:
Hey all,

Same issue here. I am running DSview v0.9.9 on Mac OS 10.12.6. Trying to run fft --> crash. 
Should I have to update the system?

[image]

Amehinou, 2018-10-06:
this bug is also on Mojave

AlmightyOatmeal, 2018-12-04:
I have the exact same issue on v0.9.9 and v0.9.8.

On my work laptop:
```
$ sw_vers
ProductName:	Mac OS X
ProductVersion:	10.13.6
BuildVersion:	17G3025
```
```
Process:               DSView [34969]
Path:                  /Applications/DSView.app/Contents/MacOS/DSView
Identifier:            com.DreamSourceLab.DSView
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           DSView [34969]
User ID:               501

Date/Time:             2018-12-03 19:52:24.718 -0600
OS Version:            Mac OS X 10.13.6 (17G3025)
Report Version:        12
Anonymous UUID:        4431C249-D158-6DB3-E494-FB4D6348CEBE

Time Awake Since Boot: 10000 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Illegal instruction: 4
Termination Reason:    Namespace SIGNAL, Code 0x4
Terminating Process:   exc handler [0]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.DreamSourceLab.DSView     	0x00000001017e23fe 0x101693000 + 1373182

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff7e65828a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff7e81f009 _pthread_wqthread + 1035
2   libsystem_pthread.dylib       	0x00007fff7e81ebe9 start_wqthread + 13

Thread 2:
0   libsystem_kernel.dylib        	0x00007fff7e65828a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff7e81f009 _pthread_wqthread + 1035
2   libsystem_pthread.dylib       	0x00007fff7e81ebe9 start_wqthread + 13

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff7e65828a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff7e81f20e _pthread_wqthread + 1552
2   libsystem_pthread.dylib       	0x00007fff7e81ebe9 start_wqthread + 13

Thread 4:: org.libusb.device-hotplug
0   libsystem_kernel.dylib        	0x00007fff7e64e20a mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff7e64d724 mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff565d6665 __CFRunLoopServiceMachPort + 341
3   com.apple.CoreFoundation      	0x00007fff565d59b7 __CFRunLoopRun + 1783
4   com.apple.CoreFoundation      	0x00007fff565d5033 CFRunLoopRunSpecific + 483
5   com.apple.CoreFoundation      	0x00007fff56613b03 CFRunLoopRun + 99
6   libusb-1.0.0.dylib            	0x0000000101c84249 0x101c79000 + 45641
7   libsystem_pthread.dylib       	0x00007fff7e81f661 _pthread_body + 340
8   libsystem_pthread.dylib       	0x00007fff7e81f50d _pthread_start + 377
9   libsystem_pthread.dylib       	0x00007fff7e81ebf9 thread_start + 13

Thread 5:
0   libsystem_kernel.dylib        	0x00007fff7e657a16 __psynch_cvwait + 10
[... 769 lines cut ...]
CoreImage                            8K        3 
CoreUI image data                   92K        3 
CoreUI image file                  404K        5 
Kernel Alloc Once                    8K        2 
MALLOC                           250.2M       39 
MALLOC guard page                   32K        7 
MALLOC_LARGE (reserved)           2188K        2         reserved VM address space (unallocated)
MALLOC_NANO (reserved)           384.0M        2         reserved VM address space (unallocated)
Memory Tag 242                      12K        2 
STACK GUARD                       56.0M       10 
Stack                             12.1M       10 
VM_ALLOCATE                        136K       18 
__DATA                            26.3M      274 
__FONT_DATA                          4K        2 
__LINKEDIT                       219.5M       34 
__TEXT                           180.0M      269 
__UNICODE                          564K        2 
mapped file                       53.6M       13 
shared memory                     3488K       11 
===========                     =======  ======= 
TOTAL                              1.2G      700 
TOTAL, minus reserved VM space   809.6M      700 
```

The fact this issue has been open for an entire year is quite concerning. This is the *one* function that I need to use; should I return the DSLogic and oscilloscope stuff?

DreamSourceLab, 2018-12-05:
@AlmightyOatmeal 
Actually we fight with this bug hardly.
Only macOS has this issue. Windows/Linux works well on FFT.
And only some of macOS computer have this issue. It's hardly to reproduce this bug.
Attached is a video about DSView running FFT on macOS 10.13 under our environment.
It's just no crash.
[DSView_FFT_on_macOS_high_sierra.zip](https://github.com/DreamSourceLab/DSView/files/2648921/DSView_FFT_on_macOS_high_sierra.zip)
Hope we can fix this bug entirely in the next version.
Sorry for any inconvenience.

AlmightyOatmeal, 2018-12-05:
@DreamSourceLab,

This issue has been open for over a year, I think you are past inconvenience. It has spanned multiple releases but yet you don't ask for any information. How can you fix it if you don't ask for information to help identify the cause?

I use valgrind, gdb, and dtrace routinely so give me something to do.

If you don't ask for help then you aren't really sorry.

shumtravi, 2018-12-06:
I think may be this bug depends of MacOS localization. I’m using it with Russian localization, don’t know maybe it helps reproduce the bug.

DreamSourceLab, 2018-12-06:
@AlmightyOatmeal 
We are glad that you want to help.
We can sent you the new DSCope hardware if this issue be fixed perfectly.

popovdg, 2019-08-19:
I have the same crash when I run DSView 0.9.9 downloaded from DreamSourceLab web side, but I have no crash in DSView which I build from master branch.