DreamSourceLab/DSView #118: Script to build the Windows version  of DSView

Cenkron, 2017-11-01:
I can build the Linux version of DSView; now I want to build the Windows version, but I can't find the build script.  I assume that like sigrok, it is done with MXE.
Can someone point me to the DSView build script for cross-compiled Windows binaries?

Cenkron, 2017-11-08:
As part of my work with respect to issue #119, I want to be able to test changes in the Windows version, as well as in the Linux version, but at the moment I don't know how to build the Windows version.
I suspect it is done either using MXE in LInux, or using MSYS2 in Windows.  I don't particularly want to reinvent the wheel, so I would greatly appreciate any help in this area.

ericjpeters, 2020-11-26:
Any updates on how to build for Windows?

olovskos, 2021-08-22:
Any updates on how to build for Windows?

RedConda, 2021-08-31:
Any updates on how to build for Windows?