DreamSourceLab/DSView #119: Triggering capability improvement

Cenkron, 2017-11-01:
I am a long time user of logic analyzers, oscilloscopes, and various other sophisticated test instruments.  I have a variety of them, many of the USB variety.
There is a lot to like about the DSLogic(s) and the DSView implementation of the host software, but there are also some things, primarily in the area of triggering, that need significant improvement.

1. Measurement of time is important, not only between events that are close to each other, but also from the trigger event to any observed event.
The latter is not well supported, primarily because the time scale is not properly synchronized to the trigger.  By definition the trigger should be at time = 0.000, within the resolution of the sampling clock.  Then, the timing of all other events is easy to read from the time scale.  This needs to be fixed.

2. It should be possible to locate the trigger point anywhere within the capture buffer, from the beginning to the end.  0 to 100%.  It is common to locate the trigger near the tail end of the capture buffer when the goal is searching in the past to find the stimulus that resulted in the triggering  event.  This typically requires the capture sampler to run continuously, and ending only after the appropriate delay from the trigger event. sufficient to fill out the capture buffer.  This may affect the design of hardware firmware.  It should be apparent that in this case, the time scale will be negative increasing toward zero for much of the capture buffer.

3. The DSView software currently repositions the capture buffer view point to the trigger point after each completed capture.  This is very undesirable.  It may well be that the area of interest within the cpture buffer is far away from the triggering event, one may run the test multiple times.  It is irritating to have to reposition the view point after running each test.  A button should however be provided to return the view point to the trigger time area when desired.  It should never happen automatically.

4. The popup window that attempts to show the completion percentage of the sampling is cute but not very useful.  In most (though not all) cases the sampling time is too short for a visual display, and it is a distraction.  It should not be used, except perhaps for sampling times that exceed a few seconds.  There should be a static display that shows the capture status of the capture buffer at all times.  This is particularly true when using repetitive captures.

5. The retrigger interval for repetitive captures needs considerably more dynamic range.  It should be possible for it to be set between 0 (meaning as small as possible), and delay times up into the multiple seconds range, completely independent of the actual capture sample rate, or of the capture cycle duration.  In the lab, there is no general rule relating the desirable magnitudes of the three time scales.

I would be happy to discuss these thoughts further with anyone that is interested.  I can posit use cases to support my view of all these issues.  It is my strong desire that the DSLogic  become the best logic analyzer in my arsenal, and I believe it has the potential to do so.

I am an embedded and real time software and hardware engineer, and I would be willing to help make this come true.

Also, I do not currently have a DScope, but I believe all of these issues are just as true for it as for the logic analyzer.  My various traditional oscilloscopes have all of these capabilities .

vatrat, 2017-11-02:
I agree with you on most points, but I think the percentage viewer is very useful. Many of the captures I do take several minutes to complete. It would definitely be helpful for the DSLogic to be transferring usb data to the computer while it's capturing data, but I don't know if it's possible with the hardware setup to capture and send data at the same time.

vatrat, 2017-11-02:
I think it would be helpful if the percentage was in the top bar or something, not the center, so that you could do other things while it was capturing.

vatrat, 2017-11-02:
Just to help with the feasibility of closing this issue and improving the software, it would probably be good to create a separate issue for each problem so that they can be closed as they're completed and in order to have more organized discussion for each issue separately.

Cenkron, 2017-11-02:
vatrat, it is a pleasure to chat with someone who understands the subject.

On splitting the issue into separate issues, I totally agree.  I will do that tomorrow, unless you would prefer to do it, especially as you have more experience with doing that in this project.
I think there is some interaction between them, but nothing major.

On the percentage monitor, I think moving it to the top bar would fully satisfy my concerns.  I seldom do such long captures, but I agree that it is useful when you need to do that.  I think I envision a colored moving progress bar, but with multiple colors to indicate the capture state, as: stopped, armed, capturing (moving), and capture complete.  If in repetitive mode, perhaps capture complete would also be moving to indicate the retrigger timeout before rearming.  In both cases, the moving aspect might be skipped if the interval is too short to waste CPU cycles on, maybe like < 0.1 Sec.  These times should be known prior to arming, and wouldn't need to be figured out on the fly.

I did not envision any change to the collection/data transmission operation.  In streaming mode I assume (but do not know for sure) that data is always being transferred, as Saleae apparently does with there low end units.  In buffered mode, I assume that collection begins when armed, treating the buffer as a circular one, and stops when there is sufficient data collected following the trigger event, then transferring all the data following the end of the collection phase.  It is probably desirable that there be a trigger holdoff until enough data has been collected to populate the buffer for the interval preceding the trigger.  I suspect that would need to be implemented in the firmware for the hardware.

If there is anything about this that I don't have right, please enlighten me.

Cenkron, 2017-11-08:
This is an update on my activities intended to improve DSView for the DSLogic(s).
I have forked DreamSourceLab/DSView to Cenkron/DSView to make my changes.
The first round of changes are now in the dev branch of that new fork.
I invite others to check it out.

Although I said I would split out the issues in my original post, I have not yet done so as I found new issues of higher priority.
The issues I have addressed in my current dev branch are:
1.  Rationalize the value legends to follow the traditional 1,2,5 sequence, rather than the somewhat ad  hoc algorithm currently employed.  This includes improvements to the minor division ticks and makes it much easier to make quantitative time measurements on the capture.
2.  Default the zoom (left and right arrow keys) to perform the zoom around the current location of the hover cursor whenever possible.  This makes it much easier to zoom in on an area of interest without constant repositioning between zoom steps.
3.  Don't reset the view to the neighborhood of the cursor following each capture, so that it is possible to position and zoom the view to an area of interest following the first capture, and not have to repeat finding this area after each successive recapture.  I have found this to be the most commonly desired use case, both in the single capture mode, and in the repetitive capture mode.

I believe these changes probably should be extended to the DScope, but I have limited them to the DSLogic(s) in this version, primarily because I don't have a DScope to test any changes against. 

If anyone cares to test my version, I am open to suggestions.  In the absence of any suggestions for further improvements, I will consider these issues fixed.

It would probably be desirable to add a mechanism to manually reset the view to the trigger point neighborhood for use whenever that is desired by the user.

I plan further changes along the lines of the issues I mentioned in my original post.

dimitarpenev, 2017-11-24:
Hello,
Can you please post link to your Cenkron/DSView repository?

Cenkron, 2017-11-24:
https://github.com/Cenkron/DSView
You want the dev branch.

DreamSourceLab, 2018-07-07:
@Cenkron Thank you very much for your constructive suggestions. We are sorry for no more interactions with customers like you before. We will pay more time on this. Most of your suggestions should be added into our Todo list.