DreamSourceLab/DSView #120: Support for SDIO

ginkgm, 2017-11-16:
could you please add support for SDIO protocol?

1. resolve CMD52, CMD53 and R5.
2. fix the bug resolving of Response by sequence rather than the bit field.

ginkgm, 2017-11-16:
Would like to contribute but there's no env for windows. pitty.

vatrat, 2018-02-04:
@gingkm Aren't the decoders written in python? I'm not too sure what you mean by being unable to contribute on Windows. I haven't tried to write one, so I'm not sure, though.

ginkgm, 2018-02-05:
@vatrat great, thanks

vatrat, 2018-02-05:
As far as I can tell, they're in `DSView/libsigrokdecode4DSL/decoders/`. I don't know if DSView will automatically detect this or not. Maybe this is a clone of sigrok functionality? Maybe look there as a start?

vatrat, 2018-02-05:
[](https://sigrok.org/wiki/Protocol_decoders)
[](https://sigrok.org/wiki/Libsigrokdecode)