DreamSourceLab/DSView #121: Suggestions: Intensity grading, Programmable Math Functions, X-Y Mode and more.

gamerpaddy, 2017-12-10:
I got a few things i would like to see in a future version since i need some of them somtimes. 

-Intensity grading display, just like a analog oscilloscope where Intensity = time the trace stays on the position. used by high end oscilloscopes.  just a software thing.

-Math functions. Right now only FFT seems to be included.  A-B  A+B and so on would be helpful. Better would be a field where you can enter your formula.  Like "=A-B/2"

-X-Y Mode where Channel 1 (2, since 1 is 0???) is Y instead of timebase.

-Touch / Tablet compatibility.  Just a option to scale some interaction elements. hard to use on a mobile tablet without mouse. The scrollbar on the bottom is tiny, impossible to hit this with a finger on 10inch tablets.

-Momentum when scrolling the waveform display. Also needed on touch sometimes.  Take a look at Saleae logic software, thats a nice scrolling effect.

-Ability to change Waveform colors.

-Longer timebase up to 1min per div.

-Dragable cursors, i cant use them without mouse / keyboard.

-More informations about the current waveform.  Like RMS, Vpp (not sure if its allready in)  Duty cycle, rise and fall time.    Well everything you could think off.

-two finger timebase zoom.

-Rough waveform preview on top bar.

-Signed drivers for DScope on win8 and higher.  Its annoying to switch into "disable driver signing" everytime i start up my pc, since i use programs too that require signing enabled.  

Or is this project allready dead?

DreamSourceLab, 2018-07-07:
Thanks for your suggestions. some of them already be supported. Others will also be added into the Todo list. More information, you can refer to the documents on our website:
http://dreamsourcelab.com/download.html

_-Momentum when scrolling the waveform display. Also needed on touch sometimes. Take a look at Saleae logic software, thats a nice scrolling effect._
-- Already support on LA mode

_-Ability to change Waveform colors._
--Already support

_-More informations about the current waveform. Like RMS, Vpp (not sure if its allready in) Duty cycle, rise and fall time. Well everything you could think off._
--RMS and Vpp support, Duty cycle need manual measurement. 

_-Signed drivers for DScope on win8 and higher. Its annoying to switch into "disable driver signing" everytime i start up my pc, since i use programs too that require signing enabled._
-New DSCope devices already support singed drivers.

DreamSourceLab, 2019-09-10:
New support in v1.00
-Math functions. Right now only FFT seems to be included. A-B A+B and so on would be helpful. Better would be a field where you can enter your formula. Like "=A-B/2"
--support Add/Subtract/Multiply/Divde

-X-Y Mode where Channel 1 (2, since 1 is 0???) is Y instead of timebase.
--support x-y mode lissajous figure

-More informations about the current waveform. Like RMS, Vpp (not sure if its allready in) Duty cycle, rise and fall time. Well everything you could think off.
-- 20 kinds of measurement (	Frequency / Period / +Duty /- Duty / +Count
Rise / Fall / +Width / -Width / BrstW
Amplitude / High / Low / RMS / Mean
Pk-Pk / Max / Min / +Over / -OVer)