DreamSourceLab/DSView #122: nRF24L01(+) decoder not working

lpdunwell, 2017-12-13:
i assume you changed something on the actual spi decoders.
this seems to work, re-checking it might be a good idea anyway.
it replaces the decode function of the pd.py

`    def decode(self, ss, es, data):
        if not self.requirements_met:
            return

        ptype, data1, data2 = data

        if ptype == 'TRANSFER':
            if self.cmd:
                # Check if we got the minimum number of data bytes
                # after the command byte.
                if len(self.mb) < self.min:
                    self.warn((ss, ss), 'missing data bytes')
                elif self.mb:
                    self.finish_command((self.mb_s, self.mb_e))

            self.next()
            self.cs_was_released = True
        elif ptype == 'DATA':
            mosi, miso = data1, data2
            pos = (ss, es)

            if miso is None or mosi is None:
                self.requirements_met = False
                raise ChannelError('Both MISO and MOSI pins required.')

            if self.first:
                self.first = False
                # First MOSI byte is always the command.
                self.decode_command(pos, mosi)
                # First MISO byte is always the status register.
                self.decode_register(pos, self.ann_reg, 'STATUS', [miso])
            else:
                if not self.cmd or len(self.mb) >= self.max:
                    self.warn(pos, 'excess byte')
                else:
                    # Collect the bytes after the command byte.
                    if self.mb_s == -1:
                        self.mb_s = ss
                    self.mb_e = es
                    self.mb.append((mosi, miso))`

lmao @ github's 'code' formatting.
anyway, it's a simple fix, and i really wonder why nobody even noticed this was broken.

DreamSourceLab, 2018-07-07:
Thanks.
It seems this protocol is rarely used.

DoctorZIP, 2019-09-01:
> Thanks.
> It seems this protocol is rarely used.

Well, could you please fix that? One of reasons to use DSLogic Plus and DSView is to debug NRF24L01 radiomodules. Please fix it in upcoming release.

DreamSourceLab, 2019-09-10:
@DoctorZIP 
Please try v1.00

closed 2019-11-11 by DreamSourceLab