DreamSourceLab/DSView #124: DSView doesn't see device on macOS High Sierra

gotoplanb, 2017-12-28:
Device LED stay amber after launching DSView on macOS 10.13.2

I can't find any info in the device manual, dreamsourcelab website, wiki, or GitHub related to connection troubleshooting. Please help.

cire831, 2017-12-28:
have you looked for the DSView showing up in a USB listing?

gotoplanb, 2017-12-28:
@cire831 I don't see a USB device that is obviously the DSLogic Pro. There's a sound device that I don't recognize, but this might be something related to the Thunderbolt hub I have. Note: I've been connecting the DSLogic Pro directly into an onboard USB port and not routing through any hubs.

[image]

cire831, 2017-12-28:
do have any straight usb ports?

do you have a linux box you can plug the dsview into?

that way you do lsusb to see if it shows up

first thing to do is see if the device is alive.  

From cPhone

v0idv0id, 2018-04-18:
Looks like macOS High Sierra blocks the USB device of the DSView...same here. The same issue is with other USB devices like arduino nanao, ws8266 ,... ;-(

phreaknik, 2018-05-04:
I am experiencing the same issue. Is there a fix to this?

DreamSourceLab, 2018-07-08:
We just release v0.99 version of DSView on macOS.
And already checked on macOS High Sierra.
It works well.
http://dreamsourcelab.com/download/DSView-v0.99.dmg

closed 2019-09-10 by DreamSourceLab