DreamSourceLab/DSView #125: ERROR: config files don't exist.

spanceac, 2018-01-16:
I'm using Ubuntu based Linux Mint.
After compiling, when trying to run DSView, the following error is thrown:
$ ./DSView
 ERROR: config files don't exist.

This patch seems to fix the problem:
diff --git a/DSView/main.cpp b/DSView/main.cpp
index 2d2a000..fbbb78e 100644
--- a/DSView/main.cpp
+++ b/DSView/main.cpp
@@ -115,10 +115,7 @@ int main(int argc, char *argv[])
 
     // Initialise DS_RES_PATH
     QDir dir(QCoreApplication::applicationDirPath());
-    if (dir.cd("..") &&
-        dir.cd("share") &&
-        dir.cd(QApplication::applicationName()) &&
-        dir.cd("res")) {
+    if (dir.cd("res")) {
         QString res_dir = dir.absolutePath() + "/";
         QByteArray str_tmp = res_dir.toLocal8Bit();
         strcpy(DS_RES_PATH, str_tmp.data());

spanceac, 2018-01-16:
Now I see that if I install the binary with "make install", then launching DSView from the default PATH works OK.
It would be good for the DSView binary to have the path to "res" folder as an argument, in case someone can't install it on their system(no permission), but wants to run it from the installation folder.

DreamSourceLab, 2018-07-07:
Thanks for your suggestion.