DreamSourceLab/DSView #130: Double window borders on Arch Linux

abougouffa, 2018-03-27:
I have an issue with the DSView GUI, the window contains its own borders (Title bar, close, maximize and minimize buttons), but the default window border of the window manager is shown, causing the GUI to have a double window border.

[image]

eldarkg, 2018-04-01:
#131 Fixed this issue. Tested on Ubuntu 17.10

abougouffa, 2018-04-01:
Yep, I have tested it on Arch Linux also, it works fine, waiting for it to be merged

abougouffa, 2018-04-02:
@DreamSourceLab Can you merge this pull request please?

DreamSourceLab, 2018-05-27:
Thanks! already merged.

abougouffa, 2018-06-04:
You are welcome, keep up the awesome work!

closed 2018-06-04 by abougouffa