DreamSourceLab/DSView #132: Using the latest libsigrokdecode

eldarkg, 2018-04-01:
Why DSView doesn't use the latest libsigrokdecode?

eldarkg, 2018-04-07:
In progress

eldarkg, 2018-04-15:
[libsigrokdecode modified fork 4DSL](https://github.com/eldarkg/libsigrokdecode4DSL).
Supports some new protocols:
```
Installing 81 protocol decoders:
0-i2c 0-spi 0-uart 1-i2c 1-spi 1-uart ade77xx adf435x adns5020 am230x arm_etmv3 
arm_itm arm_tpiu aud avr_isp can common dcf77 dmx512 ds1307 ds28ea00 dsi edid 
eeprom24xx eeprom93cxx em4100 em4305 gpib guess_bitrate i2c i2cdemux i2cfilter 
i2s ir_nec ir_rc5 jitter jtag jtag_stm32 lm75 lpc max7219 maxim_ds28ea00 mdio 
midi mlx90614 modbus mrf24j40 mxc6225xu nrf24l01 nunchuk onewire_link 
onewire_network pan1321 parallel ps2 pwm qi rfm12 rgb_led_spi rgb_led_ws281x 
rtc8564 sdcard_sd sdcard_spi spdif spi spiflash ssi32 stepper_motor swd t55xx 
tca6408a timing tlc5620 uart usb_packet usb_power_delivery usb_request 
usb_signalling wiegand xfp z80
```
Updated up to before PD v3 decoder API.

abougouffa, 2018-05-25:
Just tested it on Arch Linux, works perfectly

wtywtykk, 2018-12-30:
Some decoders are out of date and contain bug. The latest libsigrokdecode has fixed them. Is it possible to update to the latest version?
有些解码器（我发现的有dmx512）有bug。libsigrokdecode已经修复了，可以升级到最近的版本么？

DreamSourceLab, 2019-09-10:
Updated to PD v3 in v1.00

closed 2019-11-11 by DreamSourceLab

shall1234, 2021-05-21:
I specified RTC8564 as the protocol for I2C in DSView but it doesn't decode the data.  I just installed the files a couple of days ago from github, so I should be using the newest version.  What might I being doing wrong?