DreamSourceLab/DSView #139: DsLogic Pro not detected by software on Windows 7 64bit

whinis, 2018-06-27:


whinis, 2018-06-27:
Tested on a version compiled off of this github on a linux server and it recognizes the device but not the software on my windows machine.

whinis, 2018-06-28:
I have now went through a fair amount of working getting it to properly compile on windows through MSYS2 and CLIon. The build compiles and detects the DSLogic Pro properly however I cannot actually get any results as `libusb_get_pollfds` returns NULL every time.  I still am unsure as to why every windows built version does not detect my device

whinis, 2018-07-03:
It appears to be a problem with an outdated libusb such that it doesn't properly detect the usb ports on threadripper cpus

DreamSourceLab, 2018-07-07:
Hi whinis,

Please check your Device Manager of Windows 7.
If your DSLogic Pro have been recognized as Unknow Device
You can install the driver(driver path $DSView Install path$/Drivers) manually.

whinis, 2018-07-07:
I can confirm it is detected and drivers installed however if the device is connected via my motherboards usb 3.0 ports the program cannot see the device. I compiled and debugged on windows to confirm.

I purchased a Buffalo USB3.0 IFC-PCIE2U3S2 controller card originally intending for VMware use so that I could pass the device to a linux vm as Vmware also does not properly handle the device on my motherboards ports. If the device is plugged into this controller card while in windows 7 its detected by both the software and the computer in Device Manager.

As another test I downloaded a source version of libusb and compiled DSView against the source version and I got DSview recognizing the device however could not acquire likely due to a library incompatibility.

DreamSourceLab, 2018-07-08:
Is it caused by this reason?
https://github.com/libusb/libusb/wiki/Windows
"IMPORTANT: NEC/Renesas uPD720200/uPD720200A USB 3.0 users, please upgrade your drivers to version 2.1.16.0 or later. uPD720201/uPD720202 users should use version 3.x (3.0.23 or later preferred). Older versions of the driver have a bug that prevents libusb from accessing devices."

whinis, 2018-07-09:
I am currently using the MSI X399 Carbon Pro with the 1900x Threadripper.

DreamSourceLab, 2018-07-09:
Does it happen on all usb ports of your motherboards? or one dedicated port?
Additionally, you can have a try with the new DSView v0.99 version.
[http://dreamsourcelab.com/download.html](url)

whinis, 2018-07-09:
the DSView v0.99 doesn't work with any of my motherboard ports, even those labeled as usb 2.0 ports, however does work on the ports from the Buffalo USB3.0 IFC-PCIE2U3S2

whinis, 2018-07-10:
Here is it plugged into the motherboards usb 2.0 ports
[image]
[image]

here is the motherboards usb 3.0 ports
[image]
[image]

finally here is it plugged into the Renesas controller where it actually works
[image]
[image]

SomeMaker, 2019-06-14:
I have the same problem. on windows 7 and 10, on 5 different computers. it works once when dslogic is first connected

DreamSourceLab, 2019-09-10:
@SomeMaker 
It doesn't sound like a software issue.
If any device damage, please contact after-sale service.

R2D2lab, 2019-10-11:
I also have this problem.
I tried version DSView_v0.9.9_x64 and DSView_v1.0.1_x64 on Win7.
I am currently using the Gigabyte B450 AORUS PRO-CF.

On Laptop with Win10 the DSLogic works properly.

KoynovStas, 2020-09-27:
I am having the same problems with DSLogic pro. I tested DSView 0.9.9 - 1.1.2 x64 
My system: Windows 7 x64
Motherboard: ASUS Deluxe Z170
On Windows I see device like WinUSB Device but on ex info: USB-based DSL Instrument v2

If I use Kubuntu 20.04 then everything works there. On the same port. 
What should I do to make the device work in Windows 7?

SomeMaker, 2020-09-27:
I had a hardware problem.  I replaced the usb controller CY7C68013

KoynovStas, 2020-09-27:
> I had a hardware problem. I replaced the usb controller CY7C68013

On DSLogic or on Motherboard?

SomeMaker, 2020-09-27:
On DSLogic

whinis, 2020-09-27:
It's not a hardware problem and not sure if its been fixed upstream but  it cannot see the device on any usb 3.0 port on windows 7. My fix was to get IFC-PCIE2U3S2 usb card which the program can see the dslogic on.

KoynovStas, 2020-09-28:
Yes, this is a driver problem. Look: I launched VirtualBox on the same Windows installed DSView V1.12 x64 and he saw the device. The guest system in virtualbox was installed the same as on the host.
:(

KoynovStas, 2020-12-22:
I bought U3Pro16 and U3P100. In Windows 7, devices only work once. If the device was disconnected from the USB port, then DSView no longer sees the device! (Red LED). It can be plugged into other ports, or into the USB hub, or the wire can be changed. The result is zero. The device appears in the task manager. The driver is WinUSB, but DSView does not see the device. If the PC is restarted, the device can again be connected to DSView (green LED).

Someone will say that this is the reason for the USB port. or in a bad USB hub (low power). ok, then explain to me why, on the same PC. if you load into Kubuntu on 20.04. Connecting to the same port, or hub, everything works fine. I can even connect U3Pro16 first, then disconnect and enable U3P100 in the same port and everything will work. I can include both devices in one USB hub. and they both will work!
What should I do to make these devices work on Windows 7 64 bit?

Noksst, 2021-02-07:
Good day, I have the same problem. DSLogic U3Pro32 is installed on Windows 7 x64, but the LED keeps flashing. If I restart the PC, it hangs (no start) until you turn off your DSLogic U3Pro32. 
Motherboard msi b450 vdh pro max (AMD 3600). I install on this pc win10 x64 and DSLogic U3Pro32 not run. 
But on same pc win10 x64 and Intel CPU it was starts very well. 
But my programms not install on windows 10. Only Win 7 or win XP. 
I need to do work, and I'm sitting with your 32 inventing.

Propretor, 2022-03-04:
Hi!
I purchased DSLogic Plus yesterday. My work computer is Ryzen 7 3800X, ASUS Prime B350-PLUS + Windows 7 Pro.
DS Logic does not work on USB ports of the computer included in the AMD B350 chipset (VEN_1022&DEV_149C - Ryzen 3800X, VEN_1022&DEV_43BB - B350).
The external USB controller NEC uPD720201 (VEN_1912&DEV_0014) is RUNNING. The latest drivers are used. Please resolve the health issue on the AMD B350 USB ports. Thanks in advance.

KoynovStas, 2022-09-11:
> I bought U3Pro16 and U3P100. In Windows 7, devices only work once. If the device was disconnected from the USB port, then DSView no longer sees the device! (Red LED). It can be plugged into other [...]

Since version 1.2.1 this problem has been solved!

nlshipp, 2022-10-15:
I bought a U3Pro32 device and when plugged into a USB 3.0 port on my Windows 7 machine, it isn't recognized by DSView.  If I plug it into a USB 2.0 port it is recognized.  When connected to USB 3.0 port (VIA eXtensible Host Controller)  it's enumerated as a "USB-based DSL Instrument v2", but when connected to USB 2.0 port it shows up as a WinUSB Generic device.  I can go into device manager and manually update the USB3 device to WinUSB Generic device and then DSView recognizes it and the LED turns green.