DreamSourceLab/DSView #143: SPI Flash 4 wire

Tsai1005, 2018-07-12:
Support SPI Flash Dual/Quad mode decode？

vatrat, 2018-07-12:
Please show an example of what you mean. As far as I'm aware, this would be normal SPI.

Tsai1005, 2018-07-13:
[image]

DreamSourceLab, 2018-07-17:
@Tsai1005 
Could you help to provide a specification for this Dual/Quad mode of SPI?
Thanks.

Tsai1005, 2018-07-20:
[P25Q80H_datasheet_V1.1.pdf](https://github.com/DreamSourceLab/DSView/files/2212132/P25Q80H_datasheet_V1.1.pdf)
Here is one kind of flash support dual/quad mode

DreamSourceLab, 2018-07-20:
@Tsai1005 
Thanks

Tsai1005, 2019-03-12:
Dose the 4 wire SPI Flash Decoder feature support now？

DreamSourceLab, 2020-04-19:
@Tsai1005 
Please unzip attched file to #DSView#/decoders/
[qspi.zip](https://github.com/DreamSourceLab/DSView/files/4498749/qspi.zip)