DreamSourceLab/DSView #144: parallel decoders leave off last captured state by specified active edge of clk

studleylee, 2018-07-16:
The Parallel decoders leave off last captured by specified active state of clk.
They appear to stop at the last VCD change across the specified traces, but don't show the last clocked state. i.e. that last VCD(value_change_dump) value. 

Does this behavior for both "clock edge to sample on"  rising or falling selections.

DreamSourceLab, 2018-07-17:
@studleylee 
Could you help to post a screenshot to show this issue?

studleylee, 2018-07-18:
[image]

closed 2018-07-18 by studleylee

studleylee, 2018-07-18:
As I'm doing this, I understand it more: the original intent was to decode valid clocked parallel data. So it 
the data present uptol the clock edge. In my case I was using this deocder to see states of a state machine, so each clock represented a change to a new state, so I was looking at the data just after a clock, so the last post clock data is not there in this case.

So this might be a feature request to add an option to continue the data/state information in green view just past the qualified edge. At that edge the data had changed to the next state on the rising and qualified on the falling as shown in the setup.

studleylee, 2018-07-18:
Are the .py decoders compiled in at release, or can they be added by the user w/o a whole source code setup? Thanks for the great product!!!

studleylee, 2018-07-19:
Thanks for the answers. Not.

DreamSourceLab, 2018-07-19:
@studleylee 
Thanks for your comments.
No compile need for decoders. When DSView start run, all decoders will be dynamic loaded.

w2helm, 2020-01-28:
This seems like a dangling thread.  I have the same issue.  When clocked, it makes sense to render from the clock edge the data preceding that edge.  But without a clock it needs to render data from the transition forward.  Possibly just a sort of flush step when the end of the data is reached, although I think that might leave it one state behind.  It is particularly annoying when there are no transitions, because it looks like the decoder isn't working.