DreamSourceLab/DSView #145: Using the external trigger on the DSLogic Plus

kintekobo, 2018-07-26:
I don't seem to be able to find any reference to using the external trigger in any documentation. Can someone explain how this can be used please?

DreamSourceLab, 2018-07-27:
Any channel can be used as external trigger.
You just need to set your trigger condition.
Section 2.4 Trigger in our document:
http://dreamsourcelab.com/download/DSView_User_Guide_v0.99_en.pdf

kintekobo, 2018-07-27:
That I already know. I am referring to the Trigger In connection that is there along with the trigger out and clock in. The trigger in is the one I would like information on as there doesn't seem to be any way to use it in DSViw software.

DreamSourceLab, 2018-07-30:
This "TI" signal has no function now. It's reserved for compatible and further extension.

closed 2019-09-10 by DreamSourceLab

daedalus2097, 2020-11-11:
Has this changed since the last update? The CK, TI and TO signals are mentioned in the specs for the DSLogic Plus with no mention of them being reserved for future compatibility.

olovskos, 2021-04-05:
Push