DreamSourceLab/DSView #147: Serial trigger problems in V0.9.9

kintekobo, 2018-08-07:
I have been trying to get the serial trigger to work to detect the I2C bus addressing an OLED display on 0x3C without any luck. The only time the trigger will fire is if I set the data value to all 0's, or if I put a single 1 in one of the 16 positions it may trigger but the pattern triggered on is almost always not the pattern in the Data Value field. I have attached some example screen captures. As you can see the trigger point data does not match the bits set in the Data Value. Do you have any ideas what is the problem?

[image]
[image]
[image]

kintekobo, 2018-08-07:
This problem is also in 0.9.8

DreamSourceLab, 2018-08-08:
@kintekobo 
If you want to trigger when 0x3c occur in i2c bus,
The Data Bits should be set to 9 (include ack bit)
Data value should be: xxxxxxx001111000

kintekobo, 2018-08-08:
I have tried the pattern you suggested already and it does not trigger.  Below is a screen capture showing that the datastream does have the wanted pattern. And here is a link to a video capture on YouTube showing that it does not trigger.

https://youtu.be/7zRsRLaz-Gc

[image]

DreamSourceLab, 2018-08-08:
@kintekobo 
If you use "instant", all trigger flag will be ignored.
And i2c address has only 7bits, so Data value should be: xxxxxxx011110000
use "start" to make trigger work.

kintekobo, 2018-08-08:
I do realise that Instant does not use the trigger. I did it to show that there is a correctly formed packet to address 0x3C. 

In your previous message you stated that the pattern should be xxxxxxx001111000 which is what I set the trigger to for the capture on the video but you are now saying that the pattern should be xxxxxxx011110000. This pattern does trigger but it is not very intuitive and the documentation is not clear. 

In the manual you show only 8 data bits but from what I can gather it looks as if you have to include the ACK/NACK bit in the pattern for it to work. Could you perhaps look into making it clearer in the manual exactly how the pattern is matched against the data stream to create the trigger?

DreamSourceLab, 2018-08-08:
@kintekobo 
You should not set the pattern according to the decode result of i2c address. This decode result depend on your setting of shift or not shift.
You just need to check the Data Value according to the "clock flag" of your setting. 
In your case, each rising edge of channel 1 will record a data value, so the real data value is 011110000.
All of those introduction, you can find in our user guide:
https://www.dreamsourcelab.com/doc/DSView_User_Guide.pdf

closed 2019-09-10 by DreamSourceLab