DreamSourceLab/DSView #150: win 7-64 trigger setting

samnangor, 2018-10-17:
Hi There

I got problem with trigger setting; first my trigger is 90% at all time when measure at 50Mhz or faster.
In Trigger Position box shows 1% and cannot change to anything else.

DreamSourceLab, 2018-10-18:
Hi samnangor

For a hardware trigger, all data before trigger position must be stored in hardware memory before transferred to host PC. 
1. In buffer mode, which lead to a small sample depth (less sample time), you can set the position per your requirement. 
2. In stream mode, data will be transferred to PC memory in real-time for a large sample depth (long sample time). So if you fix the sample time, and select a fast sample rate, which means a larger sample depth, trigger position can only be set less than (hardware memory depth / sample depth).

samnangor, 2018-10-18:
Hi there,
Thanks for the fast response.
In buffer mode, I try to change trigger position to the 10% or whatever beside the 90% but I couldn't. Hope you can help me. The manual shows the picture of 10% and 90%, but it did not tell how to change. The trigger position in Simple Trigger has the scale 1% to 5% only. I do not know what it is for(Changing it, doesn't make any different). Here is the capture screen.
[image]

DreamSourceLab, 2018-10-20:
Hi samnangor

RLE compress is another situation. The sample deapth, which you select, may much larger than hardware depth.
You just need to know the trigger position can not be more than (hardware depth / sample depth).

samnangor, 2018-10-20:
Hi there,

Thank you very much for pointing me to the right direction.

closed 2019-09-10 by DreamSourceLab