DreamSourceLab/DSView #158: v0.9.9[DSLogic Pro] Reload *.dsl file failed

KcMeterCEC, 2018-11-28:
OS: win10
version : DSView v0.9.9 
function: DSLogic Pro
sample duration: 1 min
sample rate: 50MHz

I have saved data with *.dsl format.But when i want to reload it( **File->Open** ),the software report this error:

> Packet Error
> the content of received packet are not expected!

How can i fix this problem? Is this software bug?

DreamSourceLab, 2018-11-29:
@KcMeterCEC 
Please upload the dsl file.
No others report bug about this.
Please check the authorization of your saving path.

KcMeterCEC, 2018-11-30:
@DreamSourceLab 
I saved **dsl** file at desktop, so there is no authorization  problem.

[dsl_file.zip](https://github.com/DreamSourceLab/DSView/files/2632245/dsl_file.zip)

DreamSourceLab, 2018-11-30:
@KcMeterCEC 
It seems the file has been truncated.
Don't use desktop. 
In our test, some temporarily generated files are blocked.
We will try to find the further reason.

KcMeterCEC, 2018-12-03:
@DreamSourceLab copy that 👍

DreamSourceLab, 2019-09-10:
Please try v1.00

closed 2019-11-11 by DreamSourceLab