DreamSourceLab/DSView #159: Crash program

inf456, 2018-12-04:
When you press and hold the start button on the hot key "S" the program crashes. An interval of 1 second is insufficient. I would also like to have a timeline for multiple streaming downloads so that you can see the previous entries.

При нажатии и удерживании кнопки старт по горячей клавише "С " программа вылетает. Интервал в 1 секунду недостаточен. Так же бы хотелось иметь временную шкалу для множественных потоковых загрузок, чтобы можно было бы видеть предидущие записи.

DreamSourceLab, 2018-12-05:
Thanks.
It's an unusual bug.
By the way, we are not very clear about your second requirement. 
What do multiple streaming downloads mean?

inf456, 2018-12-16:
block capture, pause signal transmission, another block capture.

блок захвата, пауза передачи сигнала, другой блок захвата.