DreamSourceLab/DSView #160: 打开.dsl文件错误

twwscr, 2018-12-05:
提示
[image]
如图所示，保存的文件再次打开的时候就会提示这个问题，提示接收到非法格式的数据包
保存路径是放在E盘根目录下，应该不存在之前说的路径权限的问题

DreamSourceLab, 2018-12-05:
@twwscr 
麻烦提供一下有问题的dsl文件

tongwenwei, 2018-12-05:
> @twwscr
> 麻烦提供一下有问题的dsl文件

[dsl.zip](https://github.com/DreamSourceLab/DSView/files/2649006/dsl.zip)

tongwenwei, 2018-12-05:
@DreamSourceLab 
[dsl.zip](https://github.com/DreamSourceLab/DSView/files/2649007/dsl.zip)

betaguy, 2018-12-11:
@DreamSourceLab 

遇到同样的问题。我对多个DSL文件测试的结果表明，尺寸在10MB以内的DSL基本都能打开，尺寸在10MB以上的DSL部分能打开，但大部分出现了楼主一样的“接收到非法格式的数据包”提示，无法打开。

请官方尽快解决此问题。

DreamSourceLab, 2018-12-12:
@betaguy 
正在查找原因。
为了更快fix问题，也麻烦提供一下无法打开的dsl文件。

DreamSourceLab, 2019-03-21:
@tongwenwei
至今为止，发现的dsl文件打不开的问题，都是因为版本没有安装对。
之前发布过一个rc的版本，工作失误，没有打上rc的标，所以很多人以为安装的是正式版。此rc版本会导致保存的文件格式错误，所以会打不开。
建议遇到这个问题的用户，都到现在的中文官网下载正式版：
https://dreamsourcelab.cn/download/

如果安装的正式版还有问题，麻烦把问题文件上传。不然我们没法解决，因为在正式版上我们复现不了这个问题。

DreamSourceLab, 2019-03-21:
@betaguy 
你说的10M以上的dsl，我们也试了，还是无法复现问题。
可以的话，还请提供一下你那边出错的dsl文件。

closed 2019-09-10 by DreamSourceLab