DreamSourceLab/DSView #163: Edge distance measure text cut off and unselectable edges

milesfrain, 2018-12-17:
The edge distance measure feature is unreadable on HiDPI 4k displays. This is with Ubuntu 18.04 with 200% scaling. The font fits with 100% display scaling, but that can only be achieved with a global setting, and it messes up my other windows. I also tried launching with different Qt scaling settings without success.
`QT_SCREEN_SCALE_FACTORS=1 /usr/local/bin/DSView`
[image]

Here is a screenshot from the user manual for reference.
[image]

I also found that some edges are unselectable as a starting or ending point for measuring.
[image]

DreamSourceLab, 2018-12-20:
@milesfrain 
There is a display issue on HiDPI 4k displays. We will try to fix it in the next version.
Additionally, could you help to upload the dsl file of your last issue (unselectable edge)?
So we can reproduce and fix it.
Thanks.

DreamSourceLab, 2019-09-10:
Please try v1.00

milesfrain, 2019-09-10:
Excellent! This fixes both the display and edge select problems.

closed 2019-09-10 by milesfrain

milesfrain, 2019-09-10:
Found one minor issue. Might be related to this HiDPI change #228