DreamSourceLab/DSView #172: Glitch filter

DmitryD1, 2019-02-16:
It would be nice to add glitch filter in each chanel. 
And it setting in samples, ms.

DreamSourceLab, 2019-02-23:
@DmitryD1 
There are already filter setting in "options" window.
You can select "none" or "1 sample filter"

DmitryD1, 2019-02-28:
I need 2-5 sample filter.

DreamSourceLab, 2019-03-01:
@DmitryD1 
Why not slow down the sample rate?

DmitryD1, 2019-03-04:
If analiz many signal in same time with different friquency: UART 9600 and ethernet.
I need setting  programm glitch filter after read signal.
And individual setting for each channel.

doyenho, 2023-07-17:
DsView can filter glitch when capturing.
But Saleae Logic can filter when view the result, by channel, by time or samples.
[image]

Makuna, 2024-07-18:
I just hit the need for this also.
While "tuning" the capture settings could solve this, in my case a large collection of samples was taken and saved, and the glitches were not spotted in about 25% of the captures.  Having the ability to load a sample and filter it would then require not retaking samples.  In my case, time consuming and disruptive to running system that would need to be brought offline to capture them all again. 
Most of the glitches were less than 3us when a pulse is around 62ms at least.