DreamSourceLab/DSView #173: DSView crashes while adding stacked decoder

isundaylee, 2019-02-22:
DSView Version: 0.9.9
OS: macOS Mojave 10.14.3

Repro steps:

1) Set up an SPI decoder.
2) Record some SPI traffic.
3) Add a stacked decoder on top of the SPI decoder (e.g., the SPI Flash decoder).

DSView would then crash. 

Screenshot of reproduction:

[image]

Crash report (including stack trace):

```
Process:               DSView [52128]
Path:                  /Applications/DSView.app/Contents/MacOS/DSView
Identifier:            com.DreamSourceLab.DSView
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           DSView [52128]
User ID:               501

Date/Time:             2019-02-21 19:10:49.612 -0500
OS Version:            Mac OS X 10.14.3 (18D42)
Report Version:        12
Bridge OS Version:     3.3 (16P3133)
Anonymous UUID:        2A86EEF5-B6B4-0FA0-00FD-EBE40E4CA7C4

Sleep/Wake UUID:       51790C81-11A2-46C6-9C03-142415A33F59

Time Awake Since Boot: 750000 seconds
Time Since Wake:       38000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called
DSView(52128,0x10ddf35c0) malloc: *** error for object 0x60003f0959b0: pointer being freed was not allocated
 

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff6472c23e __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff647e2c1c pthread_kill + 285
2   libsystem_c.dylib             	0x00007fff646951c9 abort + 127
3   libsystem_malloc.dylib        	0x00007fff647a46e2 malloc_vreport + 545
4   libsystem_malloc.dylib        	0x00007fff647a44a3 malloc_report + 152
5   com.DreamSourceLab.DSView     	0x0000000109608a3d pv::data::decode::RowData::~RowData() + 45
6   com.DreamSourceLab.DSView     	0x0000000109607a96 0x10951e000 + 957078
7   com.DreamSourceLab.DSView     	0x0000000109607a7d 0x10951e000 + 957053
8   com.DreamSourceLab.DSView     	0x0000000109607a7d 0x10951e000 + 957053
9   com.DreamSourceLab.DSView     	0x0000000109603db8 pv::data::DecoderStack::build_row() + 56
10  com.DreamSourceLab.DSView     	0x0000000109604c9b pv::data::DecoderStack::push(boost::shared_ptr<pv::data::decode::Decoder>) + 91
11  com.DreamSourceLab.DSView     	0x00000001096116cd pv::view::DecodeTrace::on_stack_decoder(srd_decoder*) + 141
12  org.qt-project.QtCore         	0x000000010abd8160 QMetaObject::activate(QObject*, int, int, void**) + 880
13  com.DreamSourceLab.DSView     	0x0000000109631b8c pv::widgets::DecoderMenu::decoder_selected(srd_decoder*) + 60
14  com.DreamSourceLab.DSView     	0x00000001096212d5 pv::widgets::DecoderMenu::on_action(QObject*) + 117
15  org.qt-project.QtCore         	0x000000010abd8160 QMetaObject::activate(QObject*, int, int, void**) + 880
16  org.qt-project.QtCore         	0x000000010abdd40c QSignalMapper::map(QObject*) + 844
17  org.qt-project.QtCore         	0x000000010abd8160 QMetaObject::activate(QObject*, int, int, void**) + 880
[... 492 lines cut ...]
CG backing stores                 18.2M        8 
CoreGraphics                         8K        2 
CoreImage                            8K        3 
CoreServices                       172K        2 
CoreUI image data                   92K        3 
CoreUI image file                  196K        4 
Kernel Alloc Once                    8K        2 
MALLOC                             1.5G       59 
MALLOC guard page                   32K        7 
MALLOC_NANO (reserved)           384.0M        2         reserved VM address space (unallocated)
Memory Tag 242                      12K        2 
STACK GUARD                       56.0M        9 
Stack                             11.6M       10 
VM_ALLOCATE                        108K       15 
__DATA                            25.6M      272 
__FONT_DATA                          4K        2 
__LINKEDIT                       219.6M       34 
__TEXT                           177.9M      268 
__UNICODE                          564K        2 
mapped file                       50.0M       17 
shared memory                      876K       14 
===========                     =======  ======= 
TOTAL                              2.4G      719 
TOTAL, minus reserved VM space     2.0G      719 
```

DreamSourceLab, 2019-02-23:
@isundaylee 
Could you upload the *.dsl file.
It is more convenient for us to reproduce this issue.
Thanks.

isundaylee, 2019-02-23:
Here's one reproducing trace: [repro.dsl.zip](https://github.com/DreamSourceLab/DSView/files/2896572/repro.dsl.zip)

Steps to reproduce:
1. Open the `.dsl` file.
2. Set up an SPI decoder. Click OK to confirm and exit the dialog.
3. Go back into the SPI decoder dialog, and add an "SPI Flash" stacking decoder.

This seems to be an OS-specific issue - I can't seem to reproduce it on Linux.

Note that the trace file is not an "SPI Flash" trace. But this shouldn't matter. I also tested it with a custom decoder I wrote for the ENC28J60 chip, and had the same crash.

Thanks!

raszpl, 2026-02-17:
DSView saves decoder options in DSL files. If the decoder changed from the last time DSL was recorded those saved options will be  shoved into decoder regardless. Some decoders might not like values that no longer work. No idea if this is the case for the OP, but Im struggling with this right now with my custom decoder.