DreamSourceLab/DSView #182: 有几个比较影响使用的BUG

359303267, 2019-03-14:
1、40M采样的时候数据没那么准确
2、鼠标悬浮显示窗口的字体对比度比较低（蓝背景、黑字体），建议换一个对比度较大的颜色，或者由用户自定义更换颜色
3、对低分辨率屏幕的支持不好，字体发虚看不清楚。
4、居然不支持多窗口？？！
5、保存的dsl居然打不开？？？
6、软件更新速度太慢！！！！！（我们等着用呢！！）

lennonZZZ, 2019-03-20:
打不开dsl文件的问题都反映了好久了，至今没有解决！！！

DreamSourceLab, 2019-03-21:
@359303267 @lennonZZZ 
至今为止，发现的dsl文件打不开的问题，都是因为版本没有安装对。
之前发布过一个rc的版本，工作失误，没有打上rc的标，所以很多人以为安装的是正式版。此rc版本会导致保存的文件格式错误，所以会打不开。
建议遇到这个问题的用户，都到现在的中文官网下载正式版：
https://dreamsourcelab.cn/download/

如果安装的正式版还有问题，麻烦把问题文件上传。不然我们没法解决，因为在正式版上我们复现不了这个问题。

DreamSourceLab, 2019-03-21:
@359303267 
多谢反馈。
有问题麻烦每一条单独在github上开issue，便于问题追踪。

359303267, 2019-03-22:
@DreamSourceLab 
[image]  
这是那个打不开的dsl文件，偶尔就会出现这样的情况，希望尽快解决🙂
[12312.zip](https://github.com/DreamSourceLab/DSView/files/2996452/12312.zip)

animaflower, 2019-04-15:
你们这个源码是用VS2019能编译吗?

animaflower, 2019-04-15:
库都非常老了 能发个集合包吗

DreamSourceLab, 2019-09-10:
已更新v1.00版本。有问题请单条列出，便于维护。

closed 2019-09-10 by DreamSourceLab