DreamSourceLab/DSView #187: SPI protocol export packet index missing

jackshencn, 2019-04-05:
Hi,

I think this is extremely important. And this is the feature in Saleae Logic Analyzer.
By universal standard, SPI is organized into packet defined by each chip select CS# activation.

In the export and view panel, there's no such information where packet boundaries are between the bytes. Protocols usually define ChipID->ADDR->DATA in each packet. Without the packet ID, it became clueless to tell if a value is an address or data or chip ID if the data happens to match Chip ID.

Please fix this ASAP.

DreamSourceLab, 2019-09-10:
Supported spi frame decode in v1.00
Please use 1:spi, and select yes for frame decoder on setting page.

jackshencn, 2019-09-10:
Hi, Thank you! This issue is resolved!

jackshencn, 2019-09-15:
Hi, I just realized this function only works in Linux. The MacOS and Windows 1.00 and 1.01 still don't have this function. Thanks

DreamSourceLab, 2019-09-16:
@jackshencn 
All version have the same function.
Just use 1:spi, and select yes for frame decoder on setting page.

closed 2019-09-16 by jackshencn