DreamSourceLab/DSView #202: encode question:Can user add user-define encode rule to encode signal?

EdiqC, 2019-06-05:
is that user can add user-define encode rule to encode signal?
such like define channel 0 ,1,2,3 means 0000:idle 0001:read 0010:write etc

DreamSourceLab, 2019-09-10:
Yes.
Add your own decoder (written in python) to $DSView$/decoders path.

EdiqC, 2019-09-16:
i do follow your path and find some sample code there
pd.py is base on libsigrokdecode protocol decoders 

how can i decode 4 channel signal , which means state machine to me:
[channel0,channel1,channel2,channel3]=[0,0,0,0] decode 0 if means fifo empty
[channel0,channel1,channel2,channel3]=[1,0,0,0] decode 1 if means fifo half full
[channel0,channel1,channel2,channel3]=[0,1,0,0] decode 2 if means fifo full
etc.

below are my code

`class Decoder(srd.Decoder):
    api_version = 2
    id = '0:stateMachine'
    name = '0:stateMachine'
    longname = 'stateMachine for myIP'
    desc = 'Two-wire, multi-master, serial bus.'
    license = 'gplv2+'
    inputs = ['logic']
    outputs = ['stateMachine']
    channels = (
        {'id': 'gipo0', 'name': 'GPIO0', 'desc': 'GPIO0 line'},
    )
    options = (
        {'id': 'gipo1', 'name': 'GPIO1', 'desc': 'GPIO1 line'},
        {'id': 'gipo2', 'name': 'GPIO2', 'desc': 'GPIO2 line'},
        {'id': 'gipo3', 'name': 'GPIO3', 'desc': 'GPIO3 line'},
    )
    annotations = (
        ('2', 'fifoFull', 'fifofFull'),
        ('1', 'fifoHalf', 'fifoHalfFull'),
        ('0', 'fifoEty', 'fifoEmpty'),
    )
    annotation_rows = (
        #('bits', 'Bits', (7,)),
        ('addr-data', 'Address/Data', (0, 1, 2, 3, 4, 5, 6, 8, 9, 10, 11)),
        #('warnings', 'Warnings', (12,)),
    )
    binary = (

    )`