DreamSourceLab/DSView #206: The Project is DEAD?

way5, 2019-06-29:
Hello! I have both devices (logic analyzer & scope). These do work well, but another products on the market are already much more user friendly. 
Just curios if you plan to continue development
Thanks ~

DreamSourceLab, 2019-06-29:
Of course not.
We will release v1.0.0 version in the near future.
If you have any good suggestions abut being more user friendly, please feel free to let us know.

way5, 2019-07-05:
Thanks, eagerly waiting for a new release! I'll gather all info together and let you know.

dimitarpenev, 2019-08-19:
Hi Way5,

Can you point me to the DSlogic alternatives you were referring?
Thanks

huming2207, 2019-09-08:
@dimitarpenev Try PulseView. You may need to downgrade the firmware manually first.

DreamSourceLab, 2019-09-10:
v1.00 released.

closed 2019-09-10 by DreamSourceLab