DreamSourceLab/DSView #209: Oscilloscope differential measurement

fgl27, 2019-07-01:
Can oscilloscope differential measurement be added as featuring?

Like this.
[image]

Please, thanks.

DreamSourceLab, 2019-07-02:
@fgl27 This feature will be supported at the next version, which will be released in the near future.
Thanks.

DreamSourceLab, 2019-09-10:
Please try v1.00

fgl27, 2019-09-10:
Tanks helps but isn't perfect.

Because I can't tell what values the new math signal have exactly

**can the "Math Channel" be one of the channels options for the measurements?**

Put it here as a option

[image]

So I can use all of those great measurements to the "Math Channel" also

**Do I need to open a new issue just for this new request?**

Other then that the app looks great, really good improves.

DreamSourceLab, 2019-09-11:
Got it.
Just keep this issue.