DreamSourceLab/DSView #212: Suggestion: Trigger point to T=0

jnewcomb, 2019-07-08:
DSLogic: I like the platform. Good work.
Suggestion:
Can the timescale be relative to the trigger point? (Trigger point being T=0us) This means anything pre-trigger should have a negative timestamp.
Maybe it's by design? If so, I'm not using it the way it was intended!
Presently, I think the timestamp is measured from the start of the prebuffer.
Looking forward to the next release.. Thanks..

jnewcomb, 2019-07-08:
Also, every new capture resets the horizontal slide position back to the trigger point. I would prefer it not to do this. Generally, I am viewing a repeating pattern - so each time I retrigger, I have to scroll back to the position I was previously viewing. The issue with the timestamp not being T=0 does not help too!
Like I say, maybe I have not figured out the correct way to use this tool. If so, a hint would be helpful for me and others with the same question.
Thanks.

richardhendricks, 2019-09-25:
I would also like this option. This means all markers stay at their location, relative to the trigger. This would be very helpful when I want to do multiple captures and want to see if a timing spec is being violated. I can set a trigger at the rising edge, and put a marker down at -5uS and see visually when the timing is not met.