DreamSourceLab/DSView #225: error on making DSView 1.00 (ubuntu 18.04)

kasra3422, 2019-09-10:
[ 49%] Building CXX object CMakeFiles/DSView.dir/pv/view/trace.cpp.o
[ 50%] Building CXX object CMakeFiles/DSView.dir/pv/view/selectableitem.cpp.o
[ 50%] Building CXX object CMakeFiles/DSView.dir/pv/data/decoderstack.cpp.o
/home/kasra/Downloads/DSView-1.00/DSView/pv/data/decoderstack.cpp: In member function ‘void pv::data::DecoderStack::decode_data(uint64_t, uint64_t, srd_session*)’:
/home/kasra/Downloads/DSView-1.00/DSView/pv/data/decoderstack.cpp:511:40: error: invalid conversion from ‘const unsigned char**’ to ‘uint64_t {aka long unsigned int}’ [-fpermissive]
                              chunk.data(), chunk_const.data(), chunk_end - i, &error) != SRD_OK) {
                              ~~~~~~~~~~^~
/home/kasra/Downloads/DSView-1.00/DSView/pv/data/decoderstack.cpp:511:85: error: cannot convert ‘unsigned char*’ to ‘const uint8_t** {aka const unsigned char**}’ for argument ‘5’ to ‘int srd_session_send(srd_session*, uint8_t, uint64_t, uint64_t, const uint8_t**, const uint8_t*, char**)’
              chunk.data(), chunk_const.data(), chunk_end - i, &error) != SRD_OK) {
                                                                     ^
CMakeFiles/DSView.dir/build.make:1687: recipe for target 'CMakeFiles/DSView.dir/pv/data/decoderstack.cpp.o' failed
make[2]: *** [CMakeFiles/DSView.dir/pv/data/decoderstack.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/DSView.dir/all' failed
make[1]: *** [CMakeFiles/DSView.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

[image]

huming2207, 2019-09-10:
Have you tried `make clean`, then `cmake .` and `make`?

I've tested on my two devices with Manjaro. It just gets compiled and works fine

kasra3422, 2019-09-10:
I've tried 'make clean' and it didn't fix my problem.
also, I have QT5 on my system

fgl27, 2019-09-10:
Seems a gcc compiler problem the one you are using as default is more restrict, have you change the default?

Because I just installed on 18.04 all OK, I didn't change a thing regarding gcc.

kasra3422, 2019-09-10:
No, I don't think so.

[image]

so I think I need to compile it with a ubuntu live usb

fgl27, 2019-09-10:
yes that is the same gcc I have here,don't know then...

kasra3422, 2019-09-10:
I found the problem, the new libsigrokdecode4DSL wasn't installed properly, and 'srd_session_send' function on my system was the old one.

Thank you so much Felipe

closed 2019-09-10 by kasra3422